platformio/platformio-atom-ide #11: Windows problem?

IoTAdri, 2016-01-24:
THNX for the great work involved in making this package!!

As you can see in this picture something is wrong with the path in Windows (U use "\\\" instead of "\") which generates an error when Building, Uploading and Compiling.
(it works just fine by hand from the commandline)

[image]

[image]

What to do?

ivankravets, 2016-01-24:
Temporary solution: rename your `*.ino` to `*.cpp` and place `#include <Arduino.h>` at top of this file.

ivankravets, 2016-01-24:
Please upgrade PlatformIO IDE to 0.1.1

closed 2016-01-24 by ivankravets