platformio/platformio-atom-ide #14: Missing build

amittel, 2016-01-25:
Hi!
Just installed the package via Atom.
But https://atom.io/packages/build was missing.
Installed it by hand, restarted Atom and now build works.

ivankravets, 2016-01-25:
Which is your OS?

amittel, 2016-01-25:
Ah, sorry. My bad.
* Win7 pro SP1
* Atom 1.4.0
* PlatformIO IDE 0.1.0 | CLI 2.8.0.dev2

ivankravets, 2016-01-25:
Hm. Could you try installation from the scratch?

Go to "settings" and remove all packages. Then install them. Thanks.

ivankravets, 2016-01-27:
Do you have problems with the latest PlatformIO IDE?

amittel, 2016-01-27:
Since I installed the build package by hand, I had no problems in the first place. After the update everything is still working fine.
I think it was only a problem on my system.

ivankravets, 2016-01-27:
Thanks! What is your impression from the PlatformIO IDE beta? 

P.S: Don't forget to give us a start :star: https://github.com/platformio/platformio-atom-ide/stargazers

amittel, 2016-01-27:
Looking good!
Already had the inoffical predecessor installed.

ivankravets, 2016-01-27:
> Already had the inoffical predecessor installed.

Which type?

amittel, 2016-01-27:
This one: https://atom.io/packages/platomformio

closed 2016-01-27 by ivankravets