platformio/platformio-atom-ide #21: What is goto error for?

steelbrain, 2016-01-27:
I must be missing something here, because if you specify a range to linter errors, you can just click the `at line 22 col 22` part of it and it'll take you to the place of error

ivankravets, 2016-01-27:
Could you provide more info? Screnshot? Screencast? OS?

steelbrain, 2016-01-27:
@ivankravets This is the image shown in README, [image]

If you look at the message in linter panel, there at the end is `at line 22 col 3`.

If somebody clicks that, they will be taken to that file (unless it's the active one) and the exact position of the error

ivankravets, 2016-01-27:
Which is version of PlatformIO IDE?

steelbrain, 2016-01-27:
@ivankravets This is not a bug report. This is a suggestion that you don't need that goto-error panel because linter already has one that functions consistently across providers and platforms.

ivankravets, 2016-01-27:
"goto error" panel is the result of build process. Which is version of PlatformIO IDE?

steelbrain, 2016-01-27:
Like I said, this is not a bug report so the version of the package should not be important. This is just a suggestion, from a package developer, to a package developer.

Currently you are creating a completely different panel for results of build process, but there's a simpler, better way. We have a [`linter-indie`](https://github.com/atom-community/linter/wiki/Linter-Indie-API) API. If you use it in your package, you would get results from both the operations unified into a single panel. It would also reduce the complications of supporting configurations and doing DOM operations for the other panel. Furthermore, it would result in a better visual experience.

ivankravets, 2016-01-27:
There are 2 difference:

1. Linter operates with the file which was edited. User can cutomize linter with the different options than builder.
2. Builder compile WHOLE project into the single firmware. The warnings/errors can be different between linter and builder. Linter doesn't validate files which are not opened. Builder does it.

steelbrain, 2016-01-27:
Actually, linter supports messages throughout the project. All you have to do is select the `Project` tab instead of `File` tab at the bottom and it'll show all known messages from the entire project. Overall linter is very flexible.

ivankravets, 2016-01-27:
@steelbrain thanks a lot for the linter and your efforts. I mean that both instruments are useful for C/C++ developer. 

P.S: Sorry for the off-topic. Could you help us with this issue? https://github.com/nodejs/node/issues/4909 
We have disabled Serial Monitor feature in our IDE because `stdin` doesn't work correctly (or maybe is something wrong in our code).

Thanks!

ivankravets, 2016-01-27:
>  All you have to do is select the Project tab instead 

How do you know which files should be treated as "Project files"? Project can contain 20 files and only 10  will be used with "Builder".

steelbrain, 2016-01-27:
> nodejs/node#4909

We have a battle tested npm module that provides a wrapper around process execution. It also supports writing to stdin and it works well with windows too!

It's published as [`atom-linter`](http://github.com/atomlinter/atom-linter) on [npm](https://www.npmjs.com/package/atom-linter). If you were using it and wanted to spawn a process, it would be simple as this

```js
'use babel'
const Helpers = require('atom-linter')
const stdinContents = 'Something I wanna write'
Helpers.exec('python', ['/path/to/python file.python'], {stdin: stdinContents, stream: 'both'}).then(function(contents) {
  console.log('stdout', contents.stdout)
  console.log('stderr', contents.stderr)
})
```

> How do you know which files should be treated as `Project files`

Here's how it works, linter collects all of the messages it possibly can from providers who either push to it (Indie-style linters) or it pulls from (linters executed on save or on the fly). It knows which message is meant for which file using the `filePath` attribute on messages. It then only shows the user, of all the known messages, messages that are for the current, focused file.

If the user selects the `Project` tab, however, it shows all of the known messages. This does not necessarily mean that all of the project files have been linted, it means that these are all of the errors it's providers gave it after linting whatever they could.

For example, if you have two opened files with a linter like jshint or eslint (which specify [`scope`](https://github.com/atom-community/linter/wiki/Linter-API#defining-a-provider) to `file` in their provider). You'll get errors specific to that file in File tab, but if you switch to Project tab, you'll get errors from both files, but not from the files that are not open. If you close any of those files, it's messages will disappear from the project tab as the provider showing them would be text editor specific and would've been disposed.

ivankravets, 2016-01-28:
@steelbrain  Thanks a lot for the info! 

Don't hesitate to create new issue if you have any problems with PlatformIO IDE.

closed 2016-01-28 by ivankravets