platformio/platformio-atom-ide #23: build window autoclose, advice errer

sstaub, 2016-01-28:
When a build or upload is finished with **no** errors, the window is closing itself automatic.
It will be usefull to let the window stay open to see the build summaries, like memory consumption
of the board etc. I think we can close the window manual
Same problem when have errors on initialize or updating a project, when errors the red advice window closes automatic after a while, it is to fast to analyze the errors.

ivankravets, 2016-01-28:
@noseglid does Atom Build has this functionality?

sstaub, 2016-01-28:
The build window stays open when there are errors:
[image]

There is also render issue when I fold up and down the window.

ivankravets, 2016-01-28:
@sstaub thanks! Fixed in https://github.com/platformio/platformio/commit/3bc5cb0d273194cba091aa651fd9fc218ea7e4b3. Please again `Menu: PlatformIO > Upgrade PlatformIO`.

sstaub, 2016-01-28:
The info for upgrading stay no open, great work. When you can do it also for the build window...

ivankravets, 2016-01-28:
You can `Menu: PlatformIO > Toggle the build panel` and press `RUN/FIRE` button on the panel. In this case panel will not be close.

noseglid, 2016-01-28:
There is a setting in the build package where visibility can be set to `Keep visible`.

`cmd-alt-v` will also toggle the window, regardless of visibility

ivankravets, 2016-01-28:
@noseglid Thanks!!! Sorry, I missed it :(

P.S: Do you have any ideas as for https://github.com/noseglid/atom-build/issues/304#issuecomment-176294269 ?

ivankravets, 2016-01-28:
@sstaub Please go to `Atom > Preferences > Packages > Build > Settings > Panel Visibility

[image]

closed 2016-01-28 by ivankravets