platformio/platformio-atom-ide #24: gcc linter with mbed rtos.h, EthernetInterface.h

sstaub, 2016-01-28:
I have project with a mbed seeeduino ArchPro. I included some headerfiles like rtos.h etc.
The project works with no problem but the gcc linter gives me always a warning. It seems that the linter can't find the path.

[image]

ivankravets, 2016-01-28:
Try `Menu: PlatformIO > Upgrade PlatformIO` then `Menu: PlatformIO > Initialize New Project...`. Looks like I fixed it in the latest PlatformIO development build

sstaub, 2016-01-28:
When I try to Upgrade I get following error message:
[image]

ivankravets, 2016-01-28:
See https://github.com/platformio/platformio-atom-ide/issues/23#issuecomment-176328497

sstaub, 2016-01-28:
Works, thank you.

ivankravets, 2016-01-28:
May I close this issue?

ivankravets, 2016-01-28:
The linter flags are located in `.gcc-flags.json` file.

sstaub, 2016-01-28:
Yes, you can close it.

ivankravets, 2016-01-28:
All these problems are fixed in PlatformIO 2.8.0 which will be released today :) 
And.... The big surprise should be new PlatformIO IDE 0.3.0 (stable release with new features) :blush:  Please follow us on Twitter https://twitter.com/PlatformIO_Org

P.S: Please give us a stars :star:  for our projects:
* https://github.com/platformio/platformio/stargazers
* https://github.com/platformio/platformio-atom-ide

Thanks!

closed 2016-01-28 by ivankravets