platformio/platformio-atom-ide #25: linter-gcc not found

amittel, 2016-01-29:
Get this error when saving:

```
linter-gcc: Executable not found

"/usr/bin/g++" not found
```

Looks like a problem width the path. I looked into ``.platformio`` on Win, but when manually inserting a path for the linter he complains about ``Arduino.h``, but building works fine.
Tried 
* ``avr-g++.exe``
* ``arm-none-eabi-g++.exe`` 

So where should I point may PATH var for the linter?

Running
* Win7 pro
* IDE 0.2.1 | CLI 2.8.0.dev6
* linter-gcc 0.5.12

ivankravets, 2016-01-29:
Looks like you open/import old project. You need to re-init it.

amittel, 2016-01-29:
Ok. re-init worked.
So everytime the packages updates, I'll have to do a manual update to the project?

ivankravets, 2016-01-29:
No No No. The problems only with the old PlatformIO < 2.8. I will release PlatformIO 2.8.0 today and you will not have any problems in future

ivankravets, 2016-01-29:
I mean that you have to init project with IDE only 1 time.

closed 2016-01-29 by ivankravets