platformio/platformio-atom-ide #27: Find in project gives to many results

sstaub, 2016-01-29:
The new Tool-Button "Find" (in project) gives to many results. Maybe it should changed that the
function looks only in the src folder and only cpp/c/h files, but it will better that find works only
for the current buffer. Sorry, my mistake.

ivankravets, 2016-01-29:
How about Atom's comment in the 3-rd field?
[image]

sstaub, 2016-01-29:
OK, see it. But for a shortcut symbol button I think it is better to use the simple standard version.
But it is my personal opinion.

ivankravets, 2016-01-29:
This universal Find panel which allow to search everywhere within project. Don't mean something private but I don't want to make "Word" from PlatformIO IDE. I hope that users will also use Atom menu hotkeys. The idea of this toolbar to give access to PlatformIO commands/function and some global.

What do you think?

ivankravets, 2016-01-29:
Also, I implement `Menu: PlatformIO > Settings` where you can configure your request https://github.com/platformio/platformio-atom-ide/issues/23

sstaub, 2016-01-29:
That's really great.

ivankravets, 2016-02-13:
I propose to keep single universal search on the toolbar.

closed 2016-02-13 by ivankravets