platformio/platformio-atom-ide #29: Error message when updating to 0.3.2.

sstaub, 2016-01-31:
When updating to 0.3.2., the update process seems to stop, in the background i get following error message, When I restart Atom there a no more messages and it works:
[image]

orgkhnargh, 2016-01-31:
1. Which version you were upgrading from?
2. Do you see any errors in Console? (To see the Console go to `View > Developer > Toggle Developer Tools` menu item, then navigate to the Console tab inside of opened panel.)
3. Do you experience any feature regressions caused by this error? Does everything work correctly after reload?

Thanks for report!

sstaub, 2016-01-31:
1. It happens when updating from 0.3.1 to 0.3.2
3. No
updated from 0.3.2 to 0.3.3 with no errors

ivankravets, 2016-01-31:
@sstaub this issue is linked with Atom installer. I close it but please reopen if you have similar issues in future.

Thanks!

closed 2016-01-31 by ivankravets