platformio/platformio-atom-ide #30: Adding a second NodeMCU project causes build errors

tonykambo, 2016-01-31:
Hi @ivankravets , 

One NodeMCU project with Atom IDE works fine. If I add a second project both projects result in build errors that don't appear to be related to the actual source code of the project. 

My environment:
OS X 10.11.3
Atom 1.4.1
IDE 0.3.3 | CLI 2.8.2

Build errors:

`.pioenvs/nodemcu/FrameworkArduino/core_esp8266_main.cpp
xtensa-lx106-elf-gcc -o .pioenvs/nodemcu/FrameworkArduino/core_esp8266_noniso.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/core_esp8266_noniso.c
xtensa-lx106-elf-gcc: error: .pioenvs/nodemcu/FrameworkArduino/core_esp8266_noniso.c: No such file or directory
scons: *** [.pioenvs/nodemcu/FrameworkArduino/core_esp8266_noniso.o] Error 1
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/FrameworkArduino/core_esp8266_main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/core_esp8266_main.cpp`

`.pioenvs/nodemcu/libSensitiveConfig.a -Wl,--end-group
"xtensa-lx106-elf-size" -B -d .pioenvs/nodemcu/firmware.elf
text	   data	    bss	    dec	    hex	filename
217474	   2458	  31832	 251764	  3d774	.pioenvs/nodemcu/firmware.elf
"/Users/tonykambo/.platformio/packages/tool-esptool/esptool" -eo "/Users/tonykambo/.platformio/packages/framework-arduinoespressif/bootloaders/eboot/eboot.elf" -bo .pioenvs/nodemcu/firmware.bin -bm dio -bf 80 -bz 4M -bs .text -bp 4096 -ec -eo .pioenvs/nodemcu/firmware.elf -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "/Users/tonykambo/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 118, in delete_tmpcpp_file
remove(file_)
OSError: [Errno 2] No such file or directory: '/Users/tonykambo/Documents/Development/Embedded/PlatformIO/SmartLetterbox/src/tmp_ino_to.cpp'
Error in sys.exitfunc:
Traceback (most recent call last):
File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "/Users/tonykambo/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 118, in delete_tmpcpp_file
remove(file_)
OSError: [Errno 2] No such file or directory: '/Users/tonykambo/Documents/Development/Embedded/PlatformIO/SmartLetterbox/src/tmp_ino_to.cpp'
========================= [SUCCESS] Took 2.88 seconds =========================
`

Original project that used to work but now doesn't:

`-Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -Wl,-T"esp8266.flash.4m1m.ld" .pioenvs/nodemcu/src/tmp_ino_to.o -L/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/lib -L/Users/tonykambo/.platformio/packages/ldscripts -L.pioenvs/nodemcu -Wl,--start-group -lsmartconfig -lpp -lmain -lwpa -llwip -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm .pioenvs/nodemcu/libFrameworkArduinoVariant.a .pioenvs/nodemcu/libFrameworkArduino.a .pioenvs/nodemcu/libESP8266WiFi.a .pioenvs/nodemcu/libPubSubClient_ID89.a .pioenvs/nodemcu/libWire.a .pioenvs/nodemcu/libLiquidCrystal_I2C_ID576.a .pioenvs/nodemcu/libAdafruit_DHT_ID19.a .pioenvs/nodemcu/libSensitiveConfig.a -Wl,--end-group
.pioenvs/nodemcu/libPubSubClient_ID89.a: error adding symbols: Archive has no index; run ranlib to add one
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/nodemcu/firmware.elf] Error 1
.pioenvs/nodemcu/libPubSubClient_ID89.a: error adding symbols: Archive has no index; run ranlib to add one
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/nodemcu/firmware.elf] Error 1
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "/Users/tonykambo/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 118, in delete_tmpcpp_file
remove(file_)
OSError: [Errno 2] No such file or directory: '/Users/tonykambo/Documents/Development/Embedded/PlatformIO/IncubatorController/src/tmp_ino_to.cpp'
Error in sys.exitfunc:
Traceback (most recent call last):
File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "/Users/tonykambo/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 118, in delete_tmpcpp_file
remove(file_)
OSError: [Errno 2] No such file or directory: '/Users/tonykambo/Documents/Development/Embedded/PlatformIO/IncubatorController/src/tmp_ino_to.cpp'
========================= [ ERROR ] Took 3.77 seconds =========================
========================= [ ERROR ] Took 3.75 seconds =========================
3.7 sBuild failed.`

Even performing a Clean on the project results in an error. 

`Removed .pioenvs/nodemcu/ESP8266WiFi/WiFiServer.o
Removed .pioenvs/nodemcu/ESP8266WiFi/WiFiClient.o
Removed .pioenvs/nodemcu/ESP8266WiFi/WiFiUdp.o
Removed .pioenvs/nodemcu/PubSubClient_ID89/PubSubClient.o
Removed .pioenvs/nodemcu/libSensitiveConfig.a
Removed .pioenvs/nodemcu/Adafruit_DHT_ID19/DHT.o
Removed .pioenvs/nodemcu/Wire/Wire.o
Removed .pioenvs/nodemcu/libESP8266WiFi.a
Removed .pioenvs/nodemcu/libPubSubClient_ID89.a
Removed .pioenvs/nodemcu/libWire.a
Removed .pioenvs/nodemcu/libAdafruit_DHT_ID19.a
Removed .pioenvs/nodemcu/firmware.elf
Removed .pioenvs/nodemcu/firmware.bin
========================= [SUCCESS] Took 0.53 seconds =========================
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "/Users/tonykambo/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 118, in delete_tmpcpp_file
remove(file_)
OSError: [Errno 2] No such file or directory: '/Users/tonykambo/Documents/Development/Embedded/PlatformIO/SmartLetterbox/src/tmp_ino_to.cpp'
Error in sys.exitfunc:
Traceback (most recent call last):
File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "/Users/tonykambo/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 118, in delete_tmpcpp_file
remove(file_)
OSError: [Errno 2] No such file or directory: '/Users/tonykambo/Documents/Development/Embedded/PlatformIO/SmartLetterbox/src/tmp_ino_to.cpp'
========================= [SUCCESS] Took 0.53 seconds =========================`

ivankravets, 2016-01-31:
Fixed in https://github.com/platformio/platformio/commit/223a055e8b23bb03a617da52f3cb8811d49c32fe

Please go to `Settings > Packages > platformio-ide`. Uninstall it (just only this package). The install it again, it should install the latest PlatformIO 2.8.3.dev1

closed 2016-01-31 by ivankravets

tonykambo, 2016-01-31:
I performed a Clean which now works. The first project built once OK. The second project had errors. The first project is also now showing build errors:

`.pioenvs/nodemcu/FrameworkArduino/cont.o .pioenvs/nodemcu/FrameworkArduino/cont.S
xtensa-lx106-elf-gcc -o .pioenvs/nodemcu/FrameworkArduino/cont_util.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/cont_util.c
xtensa-lx106-elf-gcc: error: .pioenvs/nodemcu/FrameworkArduino/cont_util.c: No such file or directory
scons: *** [.pioenvs/nodemcu/FrameworkArduino/cont_util.o] Error 1
xtensa-lx106-elf-gcc -o .pioenvs/nodemcu/FrameworkArduino/core_esp8266_eboot_command.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -`

and the second project:

`.pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_cache.c
xtensa-lx106-elf-gcc -o .pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.c
.pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.c:23:28: fatal error: spiffs_nucleus.h: No such file or directory
#include "spiffs_nucleus.h"
^
compilation terminated.`

`.pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_nucleus.c
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/FrameworkArduino/spiffs_api.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/spiffs_api.cpp
scons: *** [.pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.o] Error 1`

tonykambo, 2016-01-31:
The error seems to vary each time you build: 

`.pioenvs/nodemcu/FrameworkArduino/spiffs_hal.o .pioenvs/nodemcu/FrameworkArduino/time.o
xtensa-lx106-elf-ar: .pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_nucleus.o: No such file or directory
scons: *** [.pioenvs/nodemcu/libFrameworkArduino.a] Error 1`