platformio/platformio-atom-ide #31: Adding a second NodeMCU project still causes build errors

tonykambo, 2016-01-31:
Hi @ivankravets, 

After platformio/platformio@223a055 in #30 errors still appear though I a now able to Clean successfully. 

`I performed a Clean which now works. The first project built once OK. The second project had errors. The first project is also now showing build errors:

.pioenvs/nodemcu/FrameworkArduino/cont.o .pioenvs/nodemcu/FrameworkArduino/cont.S xtensa-lx106-elf-gcc -o .pioenvs/nodemcu/FrameworkArduino/cont_util.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/cont_util.c xtensa-lx106-elf-gcc: error: .pioenvs/nodemcu/FrameworkArduino/cont_util.c: No such file or directory scons: *** [.pioenvs/nodemcu/FrameworkArduino/cont_util.o] Error 1 xtensa-lx106-elf-gcc -o .pioenvs/nodemcu/FrameworkArduino/core_esp8266_eboot_command.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -

and the second project:

.pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_cache.c xtensa-lx106-elf-gcc -o .pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.c .pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.c:23:28: fatal error: spiffs_nucleus.h: No such file or directory #include "spiffs_nucleus.h" ^ compilation terminated.

.pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_nucleus.c xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/FrameworkArduino/spiffs_api.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/tonykambo/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant .pioenvs/nodemcu/FrameworkArduino/spiffs_api.cpp scons: *** [.pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_check.o] Error 1`

The error seems to vary each time you build:

`.pioenvs/nodemcu/FrameworkArduino/spiffs_hal.o .pioenvs/nodemcu/FrameworkArduino/time.o xtensa-lx106-elf-ar: .pioenvs/nodemcu/FrameworkArduino/spiffs/spiffs_nucleus.o: No such file or directory scons: *** [.pioenvs/nodemcu/libFrameworkArduino.a] Error 1`

ivankravets, 2016-02-01:
PlatformIO IDE uses awesome [atom-build](https://github.com/noseglid/atom-build) package by @noseglid. However, the latest stable version of this package doesn't allow to specify unique targets within multi-projects workflow. In this case, the same "named target" calls N times (depends on the number of opened projects). 

When this functionality will be added to "atom-build", we rewrite our code to generate unique targets.

orgkhnargh, 2016-02-01:
Fix committed.

We have to wait until next atom-build release including noseglid/atom-build#309.

noseglid, 2016-02-01:
`v0.54.1` is now published with this fixed

ivankravets, 2016-02-01:
@noseglid and @orgkhnargh thanks a lot!!! :+1:

closed 2016-02-01 by ivankravets

ivankravets, 2016-02-01:
I've just tested it and new atom-build 0.54 - looks amazing with these effects! :blush: