platformio/platformio-atom-ide #33: Uncaught Error: Cannot find module 'witwip' - update failed

domschl, 2016-02-01:
[Enter steps to reproduce below:]

1. update to version 3.3 failed due to missing witwip

**Atom Version**: 1.4.1
**System**: Mac OS X 10.11.3
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v0.3.3

### Stack Trace

Uncaught Error: Cannot find module 'witwip'

```
At module.js:338

Error: Cannot find module 'witwip'
    at Module._resolveFilename (module.js:336:15)
    at Function.Module._resolveFilename (/Applications/Atom.app/Contents/Resources/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:286:25)
    at Module.require (module.js:365:17)
    at require (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:50:27)
    at getPackageJSONpath (/Users/dsc/.atom/packages/platformio-ide/node_modules/atom-package-dependencies/index.js:78:17)
    at getPackageDependencies (/Users/dsc/.atom/packages/platformio-ide/node_modules/atom-package-dependencies/index.js:87:3)
    at Object.module.exports.install (/Users/dsc/.atom/packages/platformio-ide/node_modules/atom-package-dependencies/index.js:8:5)
    at ChildProcess.<anonymous> (/Users/dsc/.atom/packages/platformio-ide/lib/install/command.js:258:44)
    at emitTwo (events.js:87:13)
    at ChildProcess.emit (events.js:172:7)
    at maybeClose (internal/child_process.js:817:16)
    at Socket.<anonymous> (internal/child_process.js:319:11)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at Pipe._onclose (net.js:469:12)
```

### Commands

```
     -0:47.9.0 settings-view:check-for-package-updates (atom-text-editor.editor.is-focused)
```

### Config

```json
{
  "core": {
    "themes": [
      "atom-dark-ui",
      "one-dark-syntax"
    ]
  }
}
```

### Installed Packages

```coffee
# User
atom-ternjs, v0.13.2
autocomplete-clang, v0.8.9
autocomplete-python, v1.5.1
build, v0.53.0
color-picker, v2.1.1
file-icons, v1.6.14
language-arduino, v0.4.1
language-cmake, v0.1.4
language-ini, v1.14.0
language-latex, v0.6.1
latex, v0.29.0
latexer, v0.2.7
linter, v1.11.3
linter-chktex, v1.2.5
linter-flake8, v1.10.0
linter-gcc, v0.5.13
linter-pep8, v1.2.1
markdown-preview-plus, v2.2.2
minimap, v4.19.0
minimap-git-diff, v4.1.8
platformio-ide, v0.3.3
travis-ci-status, v1.0.0

# Dev
No dev packages
```

ivankravets, 2016-02-01:
Hmm... Could you try to uninstall `platformio-ide` and install again?

domschl, 2016-02-01:
Deinstall / Reinstall worked without any problems.
It works now.

closed 2016-02-01 by domschl