platformio/platformio-atom-ide #34: Leonardo gcc-flags unexpected token

daniel-centore, 2016-02-02:
When using the Arduino Leonardo, the following error is produced:

[image]

```
SyntaxError: /media/DataDrive/Projects/ASME/2015-2016/Code/.gcc-flags.json: Unexpected token &
    at Object.parse (native)
    at Object.Module._extensions..json (module.js:460:27)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:50:27)
    at Object.module.exports.settings (/home/drdanielfc/.atom/packages/linter-gcc/lib/config.js:45:25)
    at Object.lint (/home/drdanielfc/.atom/packages/linter-gcc/lib/main.js:65:27)
    at promises.push.Promise.then._this.emitter.emit.linter (/home/drdanielfc/.atom/packages/linter/lib/linter-registry.js:63:26)
    at new Promise (/home/drdanielfc/.atom/packages/recent-projects/node_modules/babel-polyfill/node_modules/core-js/modules/es6.promise.js:197:7)
    at /home/drdanielfc/.atom/packages/linter/lib/linter-registry.js:62:23
    at Set.forEach (native)
    at LinterRegistry.lint (/home/drdanielfc/.atom/packages/linter/lib/linter-registry.js:60:18)
    at /home/drdanielfc/.atom/packages/linter/lib/linter.coffee:98:16
    at Function.module.exports.Emitter.simpleDispatch (/usr/share/atom/resources/app.asar/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/usr/share/atom/resources/app.asar/node_modules/event-kit/lib/emitter.js:125:28)
    at /home/drdanielfc/.atom/packages/linter/lib/editor-linter.js:30:20
    at Function.module.exports.Emitter.simpleDispatch (/usr/share/atom/resources/app.asar/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/usr/share/atom/resources/app.asar/node_modules/event-kit/lib/emitter.js:125:28)
    at TextBuffer.module.exports.TextBuffer.saveAs (/usr/share/atom/resources/app.asar/node_modules/text-buffer/lib/text-buffer.js:1067:27)
    at TextBuffer.module.exports.TextBuffer.save (/usr/share/atom/resources/app.asar/node_modules/text-buffer/lib/text-buffer.js:1037:19)
    at TextEditor.module.exports.TextEditor.save (/usr/share/atom/resources/app.asar/src/text-editor.js:602:26)
    at Pane.module.exports.Pane.saveItem (/usr/share/atom/resources/app.asar/src/pane.js:534:18)
    at Pane.module.exports.Pane.saveActiveItem (/usr/share/atom/resources/app.asar/src/pane.js:517:19)
    at Workspace.module.exports.Workspace.saveActivePaneItem (/usr/share/atom/resources/app.asar/src/workspace.js:643:35)
    at atom-workspace.commandRegistry.add.core:save (/usr/share/atom/resources/app.asar/src/register-default-commands.js:173:32)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:260:29)
    at /usr/share/atom/resources/app.asar/src/command-registry.js:3:61
    at KeymapManager.module.exports.KeymapManager.dispatchCommandEvent (/usr/share/atom/resources/app.asar/node_modules/atom-keymap/lib/keymap-manager.js:536:16)
    at KeymapManager.module.exports.KeymapManager.handleKeyboardEvent (/usr/share/atom/resources/app.asar/node_modules/atom-keymap/lib/keymap-manager.js:359:22)
    at WindowEventHandler.module.exports.WindowEventHandler.handleDocumentKeydown (/usr/share/atom/resources/app.asar/src/window-event-handler.js:97:36)
    at HTMLDocument.<anonymous> (/usr/share/atom/resources/app.asar/src/window-event-handler.js:3:61)
```

Here is the `.gcc-flags.json` the IDE produces:
```
{
  "execPath": "/home/drdanielfc/.platformio/packages/toolchain-atmelavr/bin/avr-g++",
  "gccDefaultCFlags": "-fsyntax-only -std=gnu11 -fPIC -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_LEONARDO -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_PRODUCT=\&quot;Arduino Leonardo\&quot; -DARDUINO=10607 -DPLATFORMIO=020803",
  "gccDefaultCppFlags": "-fsyntax-only -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -fPIC -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_LEONARDO -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_PRODUCT=\&quot;Arduino Leonardo\&quot; -DARDUINO=10607 -DPLATFORMIO=020803",
  "gccErrorLimit": 15,
  "gccIncludePaths": "/home/drdanielfc/.platformio/packages/framework-arduinoavr/variants/leonardo,/home/drdanielfc/.platformio/packages/framework-arduinoavr/cores/arduino,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Audio/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Bridge/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/EEPROM,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Esplora/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Ethernet/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Firmata,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/GSM/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/HID,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Keyboard/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/LiquidCrystal/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Mouse/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/RobotIRremote/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Robot_Control/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Robot_Motor/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/SD/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/SPI,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Scheduler/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Servo/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/SoftwareSerial,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/SpacebrewYun/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Stepper/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/TFT/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Temboo/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/USBHost/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/WiFi/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Wire,/home/drdanielfc/.platformio/packages/toolchain-atmelavr/avr/include,/home/drdanielfc/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include-fixed,/home/drdanielfc/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include",
  "gccSuppressWarnings": false
}
```

I was able to manually fix the file by removing the parameters that contained `\&quot;`, and things seem to be working. Here is the updated file:

```
{
  "execPath": "/home/drdanielfc/.platformio/packages/toolchain-atmelavr/bin/avr-g++",
  "gccDefaultCFlags": "-fsyntax-only -std=gnu11 -fPIC -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_LEONARDO -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DARDUINO=10607 -DPLATFORMIO=020803",
  "gccDefaultCppFlags": "-fsyntax-only -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -fPIC -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_LEONARDO -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DARDUINO=10607 -DPLATFORMIO=020803",
  "gccErrorLimit": 15,
  "gccIncludePaths": "/home/drdanielfc/.platformio/packages/framework-arduinoavr/variants/leonardo,/home/drdanielfc/.platformio/packages/framework-arduinoavr/cores/arduino,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Audio/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Bridge/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/EEPROM,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Esplora/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Ethernet/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Firmata,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/GSM/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/HID,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Keyboard/src,/home/drdanielfc/.platformio/packages [... cut ...] elfc/.platformio/packages/framework-arduinoavr/libraries/USBHost/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/WiFi/src,/home/drdanielfc/.platformio/packages/framework-arduinoavr/libraries/Wire,/home/drdanielfc/.platformio/packages/toolchain-atmelavr/avr/include,/home/drdanielfc/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include-fixed,/home/drdanielfc/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include",
  "gccSuppressWarnings": false
}
```

ivankravets, 2016-02-02:
Temporary fixed in https://github.com/platformio/platformio/commit/c3f78d19779f3a4eb45ab6c3560bc5524487f2cf

Please `Menu: PlatformIO > Upgrade PlatformIO`. The working version is `2.8.4.dev0`. 

Please reopen this issue if you still have problem.

closed 2016-02-02 by ivankravets

daniel-centore, 2016-02-03:
@ivankravets When I try to update it says:

```
You're up to date!
PlatformIO 2.8.3 is currently the newest version available.
```

ivankravets, 2016-02-08:
@daniel-centore please try to uninstall/install PlatformIO IDE package.