platformio/platformio-atom-ide #35: cannot use system installed platformio

deladriere, 2016-02-03:
I'd rather use only one install of platformio : the system one should be used by the IDE (same libs etc,)
I unchecked 'use  built-in platformIO but cannot use it
I receive : 
```
'platformio' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process.
````

ivankravets, 2016-02-03:
Which is your OS? How about to use single PlatformIO which is built-in in IDE?

deladriere, 2016-02-03:
mac osx 10.10.5
My system has already pio installed and I use Atom as editor so it was natural to test the IDE but the 2 pio installs are differents and I want to avoid that

ivankravets, 2016-02-03:
Please open PlatformIO IDE Terminal and share here screenshot of  `echo $PATH` command. 
Do you have the same output for `echo $PATH` in IDE Terminal and your native Terminal?

deladriere, 2016-02-03:
Ok re-installing gives 
```
'platformio' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process.
```

deladriere, 2016-02-03:
btw platformio-ide-terminal should install with the IDE :smile:

deladriere, 2016-02-03:
echo $path are different and now pio is not in recognized within pio-ide (use built-in is unchecked)

kommancs96, 2016-02-03:
`echo $path` gives me the following: `echo $path`

orgkhnargh, 2016-02-03:
@deladriere do you have a symlink to your system platformio somewhere in your $PATH? What does `which platformio` say?

@kommancs96 is there /bin/bash on your system? You can set a shell into the `platformio-ide-terminal` package's settings: `Preferences > Packages > platformio-ide-terminal > Settings > Shell`.

kommancs96, 2016-02-03:
@orgkhnargh No, there isn't. I use OSX. I set the shell, as you said to `usr/local/bin` , but still not working. I also get some strange js uncaught exceptions while typing.

orgkhnargh, 2016-02-04:
@kommancs96
First of all `usr/local/bin` is relative path. On my system Atom starts with [`cwd`](https://en.wikipedia.org/wiki/Working_directory) set to home directory, so `usr/local/bin` would finally resolve to `/home/orgkhnargh/usr/local/bin`. On your system it resolves to something like `/Users/kommancs96/usr/local/bin`. You've probably meant `/usr/local/bin` (note the leading slash).
Second, `/usr/local/bin` is meant to be a directory, and you can't execute a directory like it was a program. You need to set Shell to an actual shell program instead.

Please run `ls /bin/*sh` in your system terminal (not in IDE's one) to see what shells you have installed. There will probably be `/bin/sh`, which should work just fine if you set it in `platformio-ide-terminal`'s Settings.

kommancs96, 2016-02-04:
@orgkhnargh As You said, I entered `ls /bin/*sh` and got `/bin/bash	/bin/csh	/bin/ksh	/bin/sh		/bin/tcsh	/bin/zsh`. Then I tried `/bin/bash` and `/bin/sh` in `platformio-ide-terminal`'s settings, none of them worked.
I also have a problem. If I try to install platformIO shell commands, simply nothing happens. I go to `Menu-> PlatformIO -> Install Shell Commands` and nothing happens. I also tried to reinstall atom, and the packages with no success.
Is there a way to install shell commands from OSX terminal, or somehow manually?

orgkhnargh, 2016-02-04:
@kommancs96 

> I tried /bin/bash and /bin/sh in platformio-ide-terminal's settings, none of them worked.

Run `echo $SHELL` in your system terminal and set the result into platformio-ide-terminal's settings.

>  Is there a way to install shell commands from OSX terminal, or somehow manually?

Sure:

```console
sudo ln -s ~/.atom/packages/platformio-ide/penv/bin/platformio /usr/local/bin/platformio
sudo ln -s ~/.atom/packages/platformio-ide/penv/bin/pio /usr/local/bin/pio
```

And make sure that `/usr/local/bin` is in your $PATH.

ivankravets, 2016-02-05:
@orgkhnargh I've just assigned this issue to you. What need to do:

1. If users uncheck using built-in PlatformIO, then need to allow to overwrite current system PATH for IDE. 
2. Add to PlatformIO IDE new setting `PATH` (we had it before) and prefill it with the current env.PATH.
3. Use that user's specified PATH on the init state where we customize all system env vars

ivankravets, 2016-02-12:
@orgkhnargh please check if SHELL commands aren't installed properly, then need to show Error block
```
Please install shell commands manually. Open system Terminal and paste commands below:

$ sudo ln -s ~/.atom/packages/platformio-ide/penv/bin/platformio /usr/local/bin/platformio
$ sudo ln -s ~/.atom/packages/platformio-ide/penv/bin/pio /usr/local/bin/pio
```

ivankravets, 2016-02-13:
Looks like fixed in https://github.com/platformio/platformio-atom-ide/commit/f7fd6f830aa97f9b99e439d8205a578b5aa912f9

closed 2016-02-13 by ivankravets