platformio/platformio-atom-ide #37: hide/show toolbar

schmurfy, 2016-02-07:
I am trying out this project but one thing that strikes me immediately is how to only show the toolbar when it's relevant ? I was already using atom before installing this addon and this toolbar is non useful to me on non platformio projects.
I have no idea if this is possible but having an option to show it on specific projects would be nice.

ivankravets, 2016-02-07:
You can disable toolbar from settings. PlatformIO IDE package adds new item to menu `Menu: PlatformIO` where located all related commands.

schmurfy, 2016-02-07:
I saw that but I was more wondering if there was a way to do that automatically, I have no idea what the atoms' plugin api let you do thought.

PS: I love the project idea :)

ivankravets, 2016-02-25:
@schmurfy I also don't have ideas how to do it. For example, by default IDE starts without opened project. In this point, for embedded developer this Toolbar is useful, but for you is not useful because you are going to work on non-embedded project (at this time).

How to find difference?

schmurfy, 2016-02-26:
You are right, I thought about it and I am not sure there is much you can do.

ivankravets, 2016-02-26:
Please reopen this issue if you will have any propositions.

Thanks.

closed 2016-02-26 by ivankravets

eclips16, 2016-09-07:
Maybe only show the toolbar if a platformio.ini file is present in the project folder ?

ivankravets, 2016-09-07:
@eclips16 we will receive interesting glitch and bad UIX. 

If you don't need toolbar, you can disable it. All commands are duplicated in `Menu: PlatformIO`.