platformio/platformio-atom-ide #38: Compiling difference between Arduino IDE and Platformio

sanworks, 2016-02-10:
First, let me say thank you for this incredible Atom plugin!!! It majorly improves my dev time compared with the Arduino IDE!
I successfully compiled and uploaded a few programs for Arduino Due, but I ran into one bug.
The following program compiles, uploads and works with the Arduino IDE:
`char myBuffer[5] = {' ', ' ', ' ', ' ', ' '};
void setup() {}

void loop() {
  SerialUSB.println(fillMyBuffer());
  delay(1000);
}

char* fillMyBuffer() {
  for (int i = 0; i < 5; i++) {
    myBuffer[i] = 'A';
  }
  return myBuffer;
}` 

Using Atom + Platformio, I get the following build error:
error: 'fillMyBuffer' was not declared in this scope

Here's what's on the build panel:
Executing: platformio run 
[02/10/16 12:25:42] Processing dueUSB (platform: atmelsam, board: dueUSB, framework: arduino)

[02/10/16 12:25:42] Processing dueUSB (platform: atmelsam, board: dueUSB, framework: arduino)

arm-none-eabi-g++ -o .pioenvs\dueUSB\src\tmp_ino_to.o -c -fno-rtti -fno-exceptions -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DF_CPU=84000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -Dprintf=iprintf -D__SAM3X8E__ -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -DUSB_VID=0x2341 -DUSB_PID=0x003e "-DUSB_PRODUCT=\"Arduino Due\"" -DARDUINO=10606 -DPLATFORMIO=020804 -I.pioenvs\dueUSB\FrameworkArduino -I.pioenvs\dueUSB\FrameworkCMSISInc -I.pioenvs\dueUSB\FrameworkLibSam -I.pioenvs\dueUSB\FrameworkLibSam\include -I.pioenvs\dueUSB\FrameworkDeviceInc -I.pioenvs\dueUSB\FrameworkDeviceInc\3x8e\include -I.pioenvs\dueUSB\FrameworkArduinoVariant src\tmp_ino_to.cpp
arm-none-eabi-g++ -o .pioenvs\dueUSB\src\tmp_ino_to.o -c -fno-rtti -fno-exceptions -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DF_CPU=84000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -Dprintf=iprintf -D__SAM3X8E__ -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -DUSB_VID=0x2341 -DUSB_PID=0x003e "-DUSB_PRODUCT=\"Arduino Due\"" -DARDUINO=10606 -DPLATFORMIO=020804 -I.pioenvs\dueUSB\FrameworkArduino -I.pioenvs\dueUSB\FrameworkCMSISInc -I.pioenvs\dueUSB\FrameworkLibSam -I.pioenvs\dueUSB\FrameworkLibSam\include -I.pioenvs\dueUSB\FrameworkDeviceInc -I.pioenvs\dueUSB\FrameworkDeviceInc\3x8e\include -I.pioenvs\dueUSB\FrameworkArduinoVariant src\tmp_ino_to.cpp
AtomTest.ino: In function 'void loop()':
AtomTest.ino:7:32: error: 'fillMyBuffer' was not declared in this scope
AtomTest.ino: In function 'void loop()':
AtomTest.ino:7:32: error: 'fillMyBuffer' was not declared in this scope
scons: *** [.pioenvs\dueUSB\src\tmp_ino_to.o] Error 1
 [ ERROR ] Took 2.92 seconds 
scons: *** [.pioenvs\dueUSB\src\tmp_ino_to.o] Error 1
 [ ERROR ] Took 2.88 seconds 

Thanks again!

ivankravets, 2016-02-10:
Please move this issue to https://github.com/platformio/platformio/issues

closed 2016-02-10 by ivankravets

ivankravets, 2016-02-10:
@sanworks please in PlatformIO IDE `Menu: PlatformIO > Upgrade PlatformIO`

alexgorbatchev, 2017-08-06:
Same issue on 3.5.0a1 on Windows 10 with VSCode.

ivankravets, 2017-08-06:
@alexgorbatchev do you use `INO` format? Try converting it to CPP. 

* http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

alexgorbatchev, 2017-08-06:
@ivankravets nice! I actually renamed my file to `.ino` and it all compiled :) I'm just testing out a few sources from the internet and it was surprising that they didn't work at first