platformio/platformio-atom-ide #40: Modifying the MQTT_MAX_PACKET_SIZE value in PubSubClient.h (library 89) and saving it causes compile errors

tonykambo, 2016-02-14:
@ivankravets 

1. Navigate to .pioenvs/nodemcu/PubSubClinet_ID89
2. Chang value of MQTT_MAX_PACKET_SIZE from 128 to 256
3. Save header file (Cmd-S on Mac)
4. A number of GCC warnings and errors appear

Building the project does not result in the change being processed in the build.
This change has worked in the past.

The following errors are presented when saving the header file:

[image]

ivankravets, 2016-02-14:
`.pioenvs` is temporary directory for PlatformIO. Please don't edit here source files.

ivankravets, 2016-02-17:
Discussed it personally via Gitter PM.

closed 2016-02-17 by ivankravets