platformio/platformio-atom-ide #45: Pause serial monitor during (serial) upload

jbbr, 2016-02-18:
When uploading through serial connection and having serial monitor open at the same time, the upload fails. At least that's my experience with various esp8266 boards and Arduinos.
Arduino IDE disconnects serial monitor during upload and reconnects afterwards without loosing history.
It would be great to have the same behavior in PlatformIO IDE.

The current behavior should not change for OTA updates though.

ivankravets, 2016-02-23:
# Proposed implementation

- [ ] Add to PlatformIO IDE settings new setting named `Automatically close Serial Port Monitor before uploading` (checkbox, enabled by default). 
- [ ] Implement `preAtomCommandName` and `postAtomCommandName` field for each target in https://github.com/noseglid/atom-build. @orgkhnargh  please discuss it with @noseglid . 
- [ ] Attach to "Upload-based" targets (see below) `preAtomCommandName` hook. It should check if PlatformIO IDE Terminal is opened and the last command to it was issued from Serial Monitor. If yes, close it. 
- [ ] Attach to "Upload-based" targets (see below) `postAtomCommandName` hook. It should re-open Terminal with the latest Serial Terminal command and selected options.

If `Automatically close Serial Port Monitor before uploading` is disabled, then don't attach pre/post hooks.

# Upload based targets

@TODO

closed 2016-03-14 by orgkhnargh

vance, 2017-11-13:
what is the conclusion here? In latest I still have to manually close, upload, then re-open the serial monitor. is there a way to combine this into one step? If so, that should be the default behavior.

KyrychukD, 2017-11-13:
@vance there should be a checkbox called "Automatically close Serial Port Monitor before uploading" in platformio-ide package settings. Make sure that it is enabled.

mikolajzieba, 2018-01-18:
@KyrychukD Where does this setting exists in the VSC ?

ivankravets, 2018-01-18:
@mikolajzieba Atom > Settings > Packages > PlatformIO IDE

mikolajzieba, 2018-01-18:
@ivankravets Does it exists in the Visual Studio Code ?

ivankravets, 2018-01-18:
@mikolajzieba see https://github.com/platformio/platformio-vscode-ide/issues/49

commit 1f463c05 (2016-03-14): Merge branch 'feature/pause-serial-monitor'

Conflicts:
	lib/build-provider.js
	lib/config.js
	lib/serial-monitor/command.js

Resolve #45
Resolve platformio/platformio-atom-ide-terminal#6