platformio/platformio-atom-ide #47: Using built-in pio cli not possible (PATH modification fails)

jbbr, 2016-02-19:
Under certain (unknown) circumstances it's not possible to use the built-in `pio` command.

My system:
- Arch Linux
- Atom 1.5.3
- PlatformIO 2.8.4 | IDE 1.0.0
- NodeJS 5.6.0

problematic with both bash and zsh

default $PATH: `/usr/local/sbin:/usr/local/bin:/usr/bin`

$PATH is updated in "process.env.PATH":
(dev console in atom)
```
> process.env.PATH
<- "/home/user/.atom/packages/platformio-ide/penv/bin:/usr/local/sbin:/usr/local/bin:/usr/bin
```

but when clicking on e.g. serial ports toolbar button I get this:

```bash
$ pio serialports list
command not found: pio
$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/bin
$ node -e "console.log(process.env.PATH)"
/usr/local/sbin:/usr/local/bin:/usr/bin
```

orgkhnargh, 2016-02-19:
This may happen if you open a terminal too quickly and IDE is not initialized yet. Please wait few seconds after Atom start and see if `pio` is still not found.

ivankravets, 2016-02-19:
@orgkhnargh I tested this issue on @jbbr PC yesterday. This issue is linked with our Terminal. I'll explain you later.

milovanderlinden, 2016-03-04:
Same here. As described by @jbbr adding the location to the PATH will fix it. For now this is good enough for me.

ivankravets, 2016-03-04:
Guys, I forgot about temporary solution. Please click on Gear(Settings) `Packages > platformio-ide-terminal > Settings > Auto Run Command ` and specify it to `~/.atom/packages/platformio-ide/penv/bin:$PATH`

jeffcharles, 2016-04-11:
Just got bitten by this too. @ivankravets workaround works but it's one more thing I need to document for my coworkers for getting set up with platformio if we choose to use it.

ivankravets, 2016-04-11:
@jeffcharles 

>  it's one more thing I need to document for my coworkers for getting set up with platformio if we choose to use it.

That is a very specific issue. We don't know how to reproduce it. We have [more than 14k downloads](https://atom.io/packages/platformio-ide) of our IDEs and only a few users reported this problem.

We will try to reproduce it and understand how to fix it. I hope that your team's members will not have this problem :blush: 

P.S: Thank you that use PlatformIO and recommend for the team! :+1:

peterdenhaan, 2016-04-17:
I'm experiencing the same issue. Reinstalling `platformio-ide` does not help. Using Arch Linux like @jbbr - is there a pattern here? Do we all use Arch? (My Fedora box is fine.)

jclark-dot-org, 2016-04-25:
I have the same issue, and the workaround fixes it.  Not on Arch Linux; I've installed atom package platformio-ide 1.1.2, which installed platformio-ide-terminal 2.0.9 into Atom 1.7.2 running on Mac OS X 10.11.3.  `process.env.PATH` (in dev console) does include the path to the bin directory, but when a platformio terminal opens, it isn't part of the path.

kozoke, 2016-04-26:
Hi. Newbie here.  I'm getting a similar error on windows 7.  
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name,
or if a path was included, verify that the path is correct and try again.

I saw the note above about modifying the path but not sure how to do that in windows.  Any thoughts are appreciated.

ivankravets, 2016-04-26:
@kozoke could you share an output of `echo %PATH%` from PlatformIO Terminal?

kozoke, 2016-04-26:
I just figured it out.  In Windows, if you go into the environmental variables, search for Path. I added the following to the variable: C:\Users\<userid>\.atom\packages\platformio-ide\penv\Scripts

This solved the problem.  Thanks @ivankravets for offering to help!

ninjajazza, 2016-05-15:
@kozoke thanks, this fixed the same problem for me on Windows 10. I'd been getting the message "Cannot load PSReadline module." at the top of the console and spent ages seeing if that was the problem, but updating the environmental variables as you suggested fixed it for me.

closed 2016-06-28 by ivankravets

AndreKR, 2017-06-07:
Adding it to the system path sure seems to "fix" it, but it's hardly a solution as it clutters the PATH and breaks if there is a path to another command named "pio" in the PATH.

ivankravets, 2017-06-07:
http://docs.platformio.org/en/latest/faq.html#platformio-is-not-recognized-as-an-internal-or-external-command

AndreKR, 2017-06-07:
[image]

ivankravets, 2017-06-07:
@AndreKR do you use PIO IDE Terminal? See special icon on Toolbar (bottom)

AndreKR, 2017-06-07:
This terminal opened automatically when I clicked the "Platform IO: Serial Monitor" in the Platform IO toolbar.

ivankravets, 2017-06-07:
@AndreKR 
This repository is intended for PIO IDE for Atom where you posted a screenshot from VSCode.

Please open new issue here https://github.com/platformio/platformio-vscode-ide/issues

AndreKR, 2017-06-07:
Oops, sorry.

I'll do that, but you got me curious and I'll have a quick look if `updateOSEnviron()` isn't being called or if it doesn't modify `process.env.PATH` correctly. This Electron extension stuff doesn't seem too complicated.

AndreKR, 2017-06-07:
Ok, [done](https://github.com/platformio/platformio-vscode-ide/issues/15), but I don't understand what I am seeing.

raxetul, 2017-10-10:
I ve tried but does not work in 3 states:
1) only .bashrc
2) both .bashrc and atom package settings of pio
3) none

TS22082, 2018-04-22:
On Ubuntu I copy and pasted:
`export PATH=~/.atom/packages/platformio-ide/penv/bin:$PATH` 
to the auto-run command in settings and it doesn't appear to have changed anything.

I'm trying to run a serial monitor with an Arduino connected and I keep getting:
pio: command not found

It was working RIGHT before I installed and set up Nodejs for my machine.

ToPitz, 2018-08-09:
I had the same problem and for me the recommendation from [@ivankravets](https://github.com/platformio/platformio-atom-ide/issues/47#issuecomment-192302365) didn't work. I had to change the export string to a different folder:
`export PATH=~/.platformio/penv/bin:$PATH`

rwkaye, 2018-08-30:
This appears to be a major problem for a lot of people (me included) and it should be fixed, it shouldn't be a user-configuration issue.  Moreover, the advice to " export PATH=~/.atom/packages/platformio-ide/penv/bin:$PATH " is linked from all sorts of other pages here on github with nice glowing blue lines round it (as if it must be right) but this is WRONG. Moreover, there are a lot of other red herrings here in this thread, not nice for newbies. It took me a long time to scroll down to read ToPitz's post (thank you!) and get the correct path.  Please fix things so that a basic user can follow the web links and get the correct " export PATH=~/.platformio/penv/bin:$PATH " more quickly!

rwkaye, 2018-09-29:
I'm back here again.  I had to return back to remind myself what the workaround was on this issue, because after an automatic update of platformio the settings that fixed the problem were wiped by the update.  

Guys: this all looks very amateurish on your part and really needs to be sorted before any of us might want to recommend this software to others.  The problem was there since Februrary 2016 and not only hasn't been fixed but recent updates actually wipe out the workaround...

bobemoe, 2020-01-12:
@jbbr could you update the top of this ticket to point to the right comment, as @ToPitz and @rwkaye says the path is wrong. Took me a while too scroll to bottom too to work that out, but working is:

 Menu: `PlatformIO > Settings > PlatformIO IDE Terminal > Auto Run Command` and specify it to `export PATH=~/.platformio/penv/bin:$PATH`

zeruspocus, 2020-10-19:
I have the same problem, and no matter where I put the right path it will still do the same, I had to manually write the path on platformio terminal to run the command properly and compile which in my case is:
C:\Users\ZerusPocus\.platformio\penv\Scripts\platformio run -e STM32F103RC_btt_512K_USB

commit 7a2ba4a6 (2016-06-28): Export IDE's process `PATH` to the PlatformIO Terminal // Resolve #47