platformio/platformio-atom-ide #48: PlatformIO IDE hangs when installing behind a proxy.

abhishek-kakkar, 2016-02-19:
Installing PlatformIO as Atom plugin or the standalone package, I get stuck at this point:
[image]

ls -A ~/.atom/packages/platformio-ide/.cache gives only virtualenv.tar.gz but even that's 0 bytes, so nothing is getting downloaded at all.

I've configured proxy settings correctly in the Atom IDE and able to install other plugins. I have a HTTP proxy.

ivankravets, 2016-02-19:
Give me please an output of 
```
pip list
```

abhishek-kakkar, 2016-02-19:
```
~$ pip list
adium-theme-ubuntu (0.3.4)
APScheduler (3.0.1)
apt-xapian-index (0.45)
argparse (1.2.1)
beautifulsoup4 (4.3.2)
bottle (0.12.9)
ccsm (0.9.11.3)
chardet (2.0.1)
click (5.1)
cmuclmtk (0.1.5)
colorama (0.2.5)
command-not-found (0.3)
compizconfig-python (0.9.11.3)
debtagshw (0.1)
defer (1.0.6)
dirspec (13.10)
docutils (0.11)
duplicity (0.6.23)
facebook-sdk (0.4.0)
feedparser (5.1.3)
futures (3.0.3)
html5lib (0.999)
httplib2 (0.8)
Jinja2 (2.7.2)
lockfile (0.8)
lxml (3.3.3)
MarkupSafe (0.18)
mock (1.0.1)
numpy (1.8.2)
oauthlib (0.6.1)
oneconf (0.3.7.14.04.1)
openshot (1.4.3)
PAM (0.4.2)
pexpect (3.1)
Pillow (2.3.0)
pip (1.5.4)
piston-mini-client (0.7.5)
platformio (2.8.4)
PyAudio (0.2.7)
pycrypto (2.6.1)
pycups (1.9.66)
Pygments (1.6)
pygobject (3.12.0)
pyOpenSSL (0.13)
pyserial (2.6)
pysmbc (1.0.14.1)
python-apt (0.9.3.5ubuntu2)
python-dateutil (2.3)
python-debian (0.1.21-nmu2ubuntu3-linaro1)
python-mpd (0.3.0)
pytz (2014.10)
pyxdg (0.25)
PyYAML (3.10)
quantities (0-unknown)
reportlab (3.0)
requests (2.2.1)
roman (2.0.0)
semantic (1.0.3)
sessioninstaller (0.0.0)
setuptools (3.3)
six (1.5.2)
software-center-aptd-plugins (0.0.0)
Sphinx (1.2.2)
system-service (0.1.6)
Twisted-Core (13.2.0)
Twisted-Web (13.2.0)
tzlocal (1.2)
unity-lens-photos (1.0)
urllib3 (1.7.1)
wheel (0.24.0)
wsgiref (0.1.2)
wxPython (3.0.1.1)
wxPython-common (3.0.1.1)
xdiagnose (3.6.3build2)
zope.interface (4.0.5)
```
Note: I've separately installed the CLI interface as well.

ivankravets, 2016-02-19:
Please create empty directory named `penv` here `~/.atom/packages/platformio-ide/penv`. Restart IDE. Do you still see this "Installing..." Window?

abhishek-kakkar, 2016-02-19:
I now get an error, the dialog box is still there:

[image]

ivankravets, 2016-02-19:
That is good. Click `gear` icon on the PlatformIO Toolbar and go to `Packages > Installed Packages > platformio-ide > Settings` and Uncheck `Use built-in PlatformIO`. Restart IDE.

P.S: If it still doesn't work, then again goto `Packages > Installed Packages > platformio-ide > Settings` and specify `Environment PATH to run "platformio"`.

abhishek-kakkar, 2016-02-19:
I had to modify the config inside "/home/abhishek/.atom/packages/platformio-ide/lib" to stop the error message, but the "Installing..." window is still blocking my view.

ivankravets, 2016-02-19:
You don't need to touch PlatformIO IDE source code. It was my mistake. You need another empty directory: `~/.atom/packages/platformio-ide/penv/bin`.

abhishek-kakkar, 2016-02-19:
I also had to create `~/.atom/packages/platformio-ide/penv/lib` to dismiss the dialog.

Now I get:

```
abhishek@abhishek-Inspiron-7520:~/.atom/packages/platformio-ide$ platformio 
Traceback (most recent call last):
  File "/usr/local/bin/platformio", line 7, in <module>
    from platformio.__main__ import main
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 24, in <module>
    from platformio import __version__, exception, maintenance
  File "/usr/local/lib/python2.7/dist-packages/platformio/maintenance.py", line 24, in <module>
    from platformio import __version__, app, exception, telemetry
  File "/usr/local/lib/python2.7/dist-packages/platformio/app.py", line 21, in <module>
    from lockfile import LockFile
ImportError: cannot import name LockFile
```

ivankravets, 2016-02-19:
https://github.com/platformio/platformio/issues/360#issuecomment-163027538

```
sudo apt-get purge python-lockfile
sudo pip install -U platformio
```

ivankravets, 2016-02-19:
Also, I propose another solution. Manually create virtual environment for IDE.

```
rm -rf ~/.atom/packages/platformio-ide/penv
sudo pip install virtualenv
virtualenv ~/.atom/packages/platformio-ide/penv
```

Restart IDE. Go to IDE settings and revert back to use built-in PlatformIO. Restart. You will receive an error like https://github.com/platformio/platformio-atom-ide/issues/48#issuecomment-186428693. Then PlatformIO IDE `Menu: PlatformIO > Upgrade PlatformIO`. It should install the latest PlatformIO.

After it you can delete global PlatformIO CLI from OS `sudo pip uninstall platformio`.

abhishek-kakkar, 2016-02-19:
Cool, this seems to work.

So just posting the workaround in one place in case anyone comes around:
1. Install IDE / IDE plugin and close Atom
2.  
```
rm -rf ~/.atom/packages/platformio-ide/penv
sudo pip install virtualenv
virtualenv ~/.atom/packages/platformio-ide/penv
```
3. Upgrade PlatformIO from PlatformIO > Upgrade PlatformIO. Restart.

I do hope that the original issue gets fixed in the meantime. Thanks!

ivankravets, 2016-02-19:
Thanks a lot for testing! I've just updated your post https://github.com/platformio/platformio-atom-ide/issues/48#issue-134983776

orgkhnargh, 2016-02-26:
@abhishek-kakkar can you install the latest PlatofmrIO IDE from git to see if you still experience the issue?

To install the latest PlatofmrIO IDE from git run the following commands in a temrinal:

```sh
rm -r ~/.atom/packages/platformio-ide
git clone https://github.com/platformio/platformio-atom-ide ~/.atom/packages/platformio-ide
cd ~/.atom/packages/platformio-ide
npm install
```

Then start the IDE.

ivankravets, 2016-03-01:
@abhishek-kakkar please reopen issue if you still have problems

closed 2016-03-01 by ivankravets