platformio/platformio-atom-ide #50: Please, Please, Please host the files somewhere else than Sourceforge

thawkins, 2016-02-20:
I have been trying all day to download the 64-bit rpm and i have had nothing but error 500 pages, "sorry we are down for maintenance",  "sorry this project does not exist", and in most cases just "connection reset by peer" 

Its a mess, would it not be better to host the release on github, i believe they have a file download area too.

janekm, 2016-02-20:
Not to mention the whole bundling of spamware kerfuffle not too long ago, supposedly the new owners have decided to change their ways but it just feels hard to trust them now...

ivankravets, 2016-02-20:
I've just mirrored all bundles to our own DL server without CDN support. All links on the main site and docs were replaced to our DL.

Please try:
* http://platformio.org/#!/platformio-ide
* http://docs.platformio.org/en/latest/ide/atom.html#automatic-installation

ivankravets, 2016-02-20:
What is more, we already had problems with Source Forge https://twitter.com/platformio_org/status/682357992986120194. 

Now we have SERIOUSLY problems with our DL infrastructure. All PlatformIO packages are located on the SF and mirrored to our DL. If SF is off-line our DL (Europe) will be used. As result, we have received enormous interest for our PlatformIO IDE and have ~3k installs of it.

I will think on moving to some paid infrastructure:
* https://bintray.com
* Google's Cloud
* etc

Do you have any ideas where to host ~2Gb of packages with CDN?

thawkins, 2016-02-20:
Host them on Dropbox and mark them public, I belive they are backed by a
cdn. I see many small to medium websites that now do this.

That may be covered by their free offering.

ivankravets, 2016-02-20:
@thawkins Dropbox will block us. It is not allowed to for that things.

msurguy, 2016-02-24:
@ivankravets please get in touch with MaxCDN (they host BootstrapCDN.com serving terabytes of traffic each month for free). They might be able to give you free CDN since IoT is and OSS is important these days.

ivankravets, 2016-02-24:
@msurguy Are you sure that they will host 2.5Gbs of our packages? They are focused on the hosting small files.

P.S: We have mirrored all our packages to @bintray. See https://bintray.com/platformio . However, I've received form their support that they don't guarantee that we will not have problems. It seems, that they support "OSS projects" only on the paper or to attract and then "propose to switch to paid plan". Nevertheless, let's try with BinTray. I'll switch all our packages to BinTray tomorrow.

msurguy, 2016-02-24:
@ivankravets asking them wouldn't hurt :)

ivankravets, 2016-02-24:
@thawkins I've asked 1min ago :blush:  Let's wait for the answer.

msurguy, 2016-02-24:
@ivankravets you never know, maybe somebody up there would be interested in donating some of their server resources.

ivankravets, 2016-02-24:
@msurguy we are not interested for MaxCDN because PlatformIO isn't "web framework" or etc. where we will promote MaxCDN. Yep, we can place some info on the our site about MaxCDN or even BinTray but it isn't enough.

From the other side, I'm not ready to spam our users with some ads from MaxCDN or BinTray.  I'll improve PlatformIO Package Manager in PlatformIO 3.0 where each platform will contain own manifest and will have ability to specify different mirrors. In this case, if SF is off-line or file was broken form the 1-st mirror, then the next mirror will be used.

Finally, all of us understand and saw that PlatformIO grows up extremely fast. In this situation, I don't know how much bandwidth we will need through 1 month.

ivankravets, 2016-02-24:
Also, I want to mention here hint by @beriberikix as for  https://cloud.google.com/storage/

msurguy, 2016-02-24:
@ivankravets I understand. Those folks are really nice from my experience and willing to contribute (used to run Bootsnipp.com while using 100s of GBs of traffic on their CDN monthly). Who knows, they might even consider becoming a leading CDN for IoT? Usually all they ask is a link somewhere on the website and not any spammy ads.

jbaruch, 2016-02-25:
Guys, welcome to Bintray. We'll be more than happy to host 25Gb (or whatever it takes) as long as you distribute OSS software releases on Bintray. There are no limitations of file size, total storage size or bandwidth. The only requirement is having a license and a link to the source.

ivankravets, 2016-02-25:
@jbaruch Thanks a lot for proposition! I've just emailed you with the all details.

jbaruch, 2016-02-25:
@ivankravets replied.

ivankravets, 2016-02-26:
Dear Users,

All our packages have been moved to @bintray . Check https://bintray.com/platformio

To test new PlatformIO Storage in practice, please try to re-install some development platforms. For example:
```
platformio platforms list 
platformio platforms uninstall PLATFORM_NAME
platformio platforms install PLATFORM_NAME
```

P.S: @jbaruch, thanks a lot for the provided service! We hope that now our users will be satisfied with downloading :+1: :beer:

closed 2016-02-26 by ivankravets

jbaruch, 2016-02-26:
Cheers guys, enjoy!