platformio/platformio-atom-ide #51: RPM package overwrites existing Atom installation on Fedora 23

thawkins, 2016-02-20:
The  RPM package seems to be based on an older Fedora 21 1.5.3-0  package

Unfortunately when I installed it on my Fedora 23 machine it "downgraded" my current  Atom-1.5.3-1.f23.rpm package and broke my Atom install. I use Atom to edit PHP sites.

ivankravets, 2016-02-20:
You don't need to install our IDE if you have Atom already installed. Just install [PlatformIO IDE package for Atom](https://atom.io/packages/platformio-ide).

See http://docs.platformio.org/en/latest/ide/atom.html#manual-installation

P.S: Should I add to main site ( http://platformio.org/#!/platformio-ide ) near "Green" download button new button "Install Package" and forward user to https://atom.io/packages/platformio-ide?

thawkins, 2016-02-20:
Noted, I will uninstall, put back the default distro Atom package and try
that.

thawkins, 2016-02-20:
It turns out that just running 

"sudo dnf update" reinstalled the package that was replaced and, and it seems to still be working.

closed 2016-02-20 by thawkins

ivankravets, 2016-02-20:
I've just released hotfix for PlatformIO IDE 1.0.3. 

And how about 

> Should I add to main site ( http://platformio.org/#!/platformio-ide ) near "Green" download button new button "Install Package" and forward user to https://atom.io/packages/platformio-ide?