platformio/platformio-atom-ide #52: PlatformiO website donation system has problems

thawkins, 2016-02-20:
I just tried to donate  $50 to platformio to help cover the cost of file hosting.  The donation system is "strange",  I got to the externaly hosted donation page, and put in my details and cc card details,  got sent a small message,  put in the pass code,  got to the part where I fill in my personal details, and when I pressed submit, got a small red panel with an error message in Russian.  When I tried to capture the message, as soon as I moved my mouse of did anything else the panel disappeared. 

So now I don't know if my donation has been submitted or not,  and I don't know what the error was. I have also seem to have submitted a bunch of personal data to a site in Russia,  which appears to send out sms messages in Dutch.  Not very reassuring. 

I suspect you have a very low donation rate?, it may be worth taking this up with your provider.

ivankravets, 2016-02-20:
1. We are not from Russia and don't have any relation with it. We are from Ukraine. However, some people use Russian language in Ukraine and as result you have seen that message.
2. "I suspect you have a very low donation rate?". You are right :( I see red records in bank system :( I'm contacting with them to decide why they refuse donations :disappointed: 
3. "So now I don't know if my donation has been submitted or not". I don't see any records related to you. Looks like they event didn't try to deduct some from your card. 

We'll try to find other solutions with donation.

P.S: I'll answer here when receive answer from the bank.

thawkins, 2016-02-20:
I'm in the Philippines, so I know some systems refuse cards from my part of
the world, but I successfully purchase stuff from amazon etc on this card
so there should not be an issue.

sstaub, 2016-02-20:
I have the same problem as thawkins, i tried to donate €25.- but at the end of personal detail i got
also a cyrillic error message.

ivankravets, 2016-02-20:
Guys, should you pay any tax from PayPal donations in your countries? Maybe, we should ask someone to receive our donations. I propose to use that donations for DL Storage.

thawkins, 2016-02-20:
I'm prepared to send finds by Western union  or equivalent,  if you have a
service like that in Ukraine.

ivankravets, 2016-02-20:
@sstaub could you try again and provide me screenshot of that error? I'll contact with my bank.

ivankravets, 2016-02-20:
@sstaub are you from?

ivankravets, 2016-02-20:
@thawkins  They say that they received refuse from your bank :( The error message explains it in Russian :( I asked them to fix it and show further messages on the selected language.

ivankravets, 2016-02-20:
> I'm prepared to send finds by Western union  or equivalent,  if you have a
service like that in Ukraine.

Please wait for a while. I'll try to find other solutions. Really, we don't need that money as cache. What is more, you will lost a 15-20% fees. I propose to receive that money on normal bank card and than re-use it for PlatformIO needs (file storage and etc)

ivankravets, 2016-02-20:
Guys, could you try https://www.bountysource.com/teams/platformio ? They want 10% and this is OK for us.

ivankravets, 2016-02-20:
@thawkins please report if https://salt.bountysource.com/checkout/amount?team=platformio works for you I'll forward all donors to BountySource.

thawkins, 2016-02-20:
Noted, I will hold, but there is $50 here with platformio's name on it. I
love the project, I do a lot of hobby work with 3D printers building and
configuring, and rebuilding versions of Marlin firmware.

You should talk to the arduino.cc guys, now you have the IDE, they may be
interested in using it as an upgraded coding and distribution tool. Even in
its 1.0 version form it's significantly better than the existing arduino
IDE, I would love to see how far it can get polished after a few
iterations. If I was not such a lousy js programmer, I would volunteer to
help out. (I really don't understand the js scoping  rules).

ivankravets, 2016-02-20:
I've just updated that page https://salt.bountysource.com/teams/platformio . Let's if it works for a few days.

>  I love the project, I do a lot of hobby work with 3D printers building and
configuring, and rebuilding versions of Marlin firmware.

That is great! We wish you success with your projects.

> You should talk to the arduino.cc guys, now you have the IDE, they may be
interested in using it as an upgraded coding and distribution tool. Even in
its 1.0 version form it's significantly better than the existing arduino
IDE

I'm working on it and trying to find support from the hardware vendors. That is important for PlatformIO Organization to do amazing work and keep all PlatformIO Ecosystem totally free for all.

thawkins, 2016-02-20:
Re bounty source, it almost worked, I don't have a PayPal account,  I have
a problem with verifying my card with PayPal,  they take a small sum of
your card and then ask you to input the transaction number as a security
verification code, they assume that everybody has online billing, and can
get that number immediately.  However here in the philippines they don't
have online billing so PayPal helpfully says to wait a few weeks until your
paper statement comes. I will call the cc card company tomorrow and ask
them if they will email me the transaction number on my registered email
address.

ivankravets, 2016-02-20:
@thawkins thanks, keep me in touch. 

Nevertheless, I changed all links on http://platformio.org/#!/donate and now they redirect to BountySource.

sstaub, 2016-02-20:
I used the new service without problems and donated via paypal. For your info, i come from germany.

ivankravets, 2016-02-20:
@sstaub Thanks a lot! I see it. 

How about the tax as for PayPal? Should you pay something for donations?

sstaub, 2016-02-20:
I never paid taxes when donating per paypal. If I buy a product! outside EU I have to pay taxes in germany if the invoice is more than 40€.

ivankravets, 2016-02-21:
@sstaub I meant, should you pay taxes when someone donates to your card?

sstaub, 2016-02-21:
Uuups, I really don't know. I think here in germany you must pay taxes if you are a company and it is a normal income.

ivankravets, 2016-02-26:
@thawkins Please reopen this issue if you still have any problems.

closed 2016-02-26 by ivankravets

thawkins, 2016-02-26:
Noted, I still waiting for my statement so I can validate my cc against
PayPal.

thawkins, 2016-03-28:
Quick note, I finally got my CC card verified on PayPal, so i shot you a $50 donation.

ivankravets, 2016-03-28:
@thawkins Wow!!! That is record with donation :blush: Thanks a lot!!!

https://twitter.com/platformio_org/status/714414556475498496

thawkins, 2016-03-28:
> @thawkins <https://github.com/thawkins> Wow!!! That is record with
> donation [image: :blush:] Thanks a lot!!!
>
> https://twitter.com/platformio_org/status/714414556475498496

It's a very very cool tool, especially with the IDE now coming available.
It's worth much more than that.