platformio/platformio-atom-ide #54: Linter bug with user library

sstaub, 2016-02-20:
I use in my program a user library, but I get stupid linter warning message:
[image]
When I comment the include Header file out (// #include "TextLCD.h") the warning goes away.
[image]

ivankravets, 2016-02-20:
Please add `-Wno-comment` to `gccDefaultCFlags` and `gccDefaultCppFlags` fields after `-Wall` in `.gcc-flags.json` file.

Does it work?

sstaub, 2016-02-20:
Yes, it works.
Thank you for that fast replay.

ivankravets, 2016-02-20:
Should I include it by default for all projects?

sstaub, 2016-02-20:
I think yes. But I didn't really understand why this warning comes up, and the col-number is not existing.

sstaub, 2016-02-21:
Here is a copy from the code, which cause the following warning:
"gcc warning multi-line comment [-Wcomment] at line 54 col 34"
- The column doesn't exist. 
- The warning disappears when deleting #include "TextLCD.h"
- the code works, also the functionality of the user library
[image]
[image]
[mbedBase.zip](https://github.com/platformio/platformio-atom-ide/files/139535/mbedBase.zip)

ivankravets, 2016-02-21:
I opened issue https://github.com/platformio/platformio/issues/528 where you will have ability to use `build_flags = -Wno-comment -Wl,-u,_printf_float,-u,_scanf_float` and PlatformIO IDE `Menu: Rebuild C/C++ Project Index`.

Please subscribe to https://github.com/platformio/platformio/issues/528 and test when it will be fixed.

closed 2016-02-21 by ivankravets