platformio/platformio-atom-ide #55: "PlatformIO IDE: Installing ..." keeps showing up on each restart/reload of my atom.io installation

kandril, 2016-02-24:
### General issue description

My PlatformIO-IDE installation done via the currently available `platformio-atom-linux-x86_64.deb` from http://platformio.org/#!/ keeps showing me this on every restart/window reload:

[image]
---

### Additional information

Suggestions from [PlatformIO IDE hangs when installing behind a proxy. #48](https://github.com/platformio/platformio-atom-ide/issues/48) didn't help/change the _status quo_ of this issue, unfortunately.

#### Probably useful output

I'm using Xubuntu.

- Output of `uname -a`:
```
Linux anon 3.19.0-51-generic #57~14.04.1-Ubuntu SMP Fri Feb 19 14:36:55 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

- Output of `gcc --version`:
```
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.1) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

- Output of `pip list`:
```
apt-xapian-index (0.45)
argparse (1.2.1)
bottle (0.12.9)
chardet (2.0.1)
click (5.1)
colorama (0.3.6)
command-not-found (0.3)
debtagshw (0.1)
defer (1.0.6)
dirspec (13.10)
dnspython (1.11.1)
gyp (0.1)
html5lib (0.999)
httplib2 (0.8)
lockfile (0.12.2)
lxml (3.3.3)
oauthlib (0.6.1)
oneconf (0.3.7.14.04.1)
PAM (0.4.2)
pexpect (3.1)
Pillow (2.3.0)
pip (1.5.4)
piston-mini-client (0.7.5)
platformio (2.8.4)
psutil (1.2.1)
pycrypto (2.6.1)
pycups (1.9.66)
pygobject (3.12.0)
pyOpenSSL (0.13)
pyserial (3.0.1)
pysmbc (1.0.14.1)
python-apt (0.9.3.5ubuntu2)
python-debian (0.1.21-nmu2ubuntu2)
pyxdg (0.25)
qrtools (1.2)
reportlab (3.0)
requests (2.9.1)
sessioninstaller (0.0.0)
setuptools (3.3)
six (1.5.2)
software-center-aptd-plugins (0.0.0)
Twisted-Core (13.2.0)
Twisted-Web (13.2.0)
urllib3 (1.7.1)
virtualenv (14.0.6)
wheel (0.24.0)
wsgiref (0.1.2)
wxPython (2.8.12.1)
wxPython-common (2.8.12.1)
zope.interface (4.0.5)
```

orgkhnargh, 2016-02-24:
Please look at the output of `apm list | grep platformio-ide@`. If installed version is older than 1.0.4, try `apm remove platformio-ide && apm install platformio-ide`.

kandril, 2016-02-24:
The output of `apm list | grep platformio-ide@` is:
```
├── platformio-ide@1.0.4
```
So no need for `apm remove platformio-ide && apm install platformio-ide` I guess.

ivankravets, 2016-02-25:
@kandril Could you provide TeamViewer session to debug it?

kandril, 2016-02-25:
Wouldn't feel comfortable with a TeamViewer session, to be honest.
Is there any other way to provide you with additional, useful information on this issue?

ivankravets, 2016-02-25:
No problems. Please debug it manually. Add new line before https://github.com/platformio/platformio-atom-ide/blob/develop/lib/install/command.js#L525
```js
console.log(state.restored, state.envShouldBeCreated, state.packageManagementIsNecessary);
```

1. `View > Developer  > Toggle Dev Tools`
2. `View > Developer  > Reload Window`

Post here the output from the console.

kandril, 2016-02-25:
```javascript
Window load time: 1925ms
/deep/ combinator is deprecated. See https://www.chromestatus.com/features/6750456638341120 for more details.
::shadow pseudo-element is deprecated. See https://www.chromestatus.com/features/6750456638341120 for more details.
true Error: ENOENT: no such file or directory, stat '/home/kandril/.atom/packages/platformio-ide/penv/bin'(…) true
POST https://www.google-analytics.com/collect?t=event&ec=window&ea=ended&ev=3766…a8a97a7346a05a4ed96f078dd35748b9&an=atom&av=1.5.3&sr=1920x1080&aiid=stable net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=event&ec=window&ea=started&v=1&t…a8a97a7346a05a4ed96f078dd35748b9&an=atom&av=1.5.3&sr=1920x1080&aiid=stable net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=timing&utc=shell&utv=load&utt=62…a8a97a7346a05a4ed96f078dd35748b9&an=atom&av=1.5.3&sr=1920x1080&aiid=stable net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=timing&utc=core&utv=load&utt=192…a8a97a7346a05a4ed96f078dd35748b9&an=atom&av=1.5.3&sr=1920x1080&aiid=stable net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=event&ec=core&ea=activated&el=0.55.0&cid=4382ee17a8a97a7346a05a4ed96f078dd35748b9&v=1&tid=UA-47615700-5 net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=event&ec=build&ea=tool%20eligibl…tformIO&cid=4382ee17a8a97a7346a05a4ed96f078dd35748b9&v=1&tid=UA-47615700-5 net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=event&ec=deprecation-v3&ea=autoc…a8a97a7346a05a4ed96f078dd35748b9&an=atom&av=1.5.3&sr=1920x1080&aiid=stable net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=event&ec=keymap&ea=registered&el…l-alt-b&cid=4382ee17a8a97a7346a05a4ed96f078dd35748b9&v=1&tid=UA-47615700-5 net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=event&ec=keymap&ea=registered&el…l-alt-c&cid=4382ee17a8a97a7346a05a4ed96f078dd35748b9&v=1&tid=UA-47615700-5 net::ERR_SSL_PROTOCOL_ERROR
POST https://www.google-analytics.com/collect?t=event&ec=keymap&ea=registered&el…l-alt-u&cid=4382ee17a8a97a7346a05a4ed96f078dd35748b9&v=1&tid=UA-47615700-5 net::ERR_SSL_PROTOCOL_ERROR
true Error: ENOENT: no such file or directory, stat '/home/kandril/.atom/packages/platformio-ide/penv/bin'
    at Error (native) true

```

ivankravets, 2016-02-25:
Hm... Are you sure that performed https://github.com/platformio/platformio-atom-ide/issues/48#issuecomment-186442111 ?

After it, you should had PlatformIO installed. Do you use proxy? 

In any case, try close IDE and remove `/home/<user>/.atom/packages/platformio-ide/`. The start IDE.

kandril, 2016-02-25:
Done. Directory got removed entirely. Package was reinstalled automatically after an IDE restart, but the issue still stands, unfortunately

kandril, 2016-02-26:
##### For those interested in a workaround until the next IDE release in a few days:
For me, the `build` package wasn't loaded correctly (by the Editor) and caused the "PlatformIO: Installing ..." window/process to get stuck.

This was due to me having a directory called `build` in my working path (I start the Editor with `atom %U`, so my working path was `~/`), which the editor tried to load as the Editor's `build`-package, which apparently failed.

 `mv ~/build ~/build-bkp`, AKA renaming it, solved the issue for me. But it's only a workaround!

orgkhnargh, 2016-02-26:
@kandril can you install the IDE as described [here](https://github.com/platformio/platformio-atom-ide/issues/48#issuecomment-189379370), then `mv ~/build-bkp ~/build`, start Atom and check if `Installing...` panel stopped showing up on every reload/restart?

kandril, 2016-02-26:
@orgkhnargh Works like a charm, now. Thanks for your effort :)

closed 2016-03-01 by ivankravets

saschaludwig, 2017-05-20:
I have the same issue after updating. PlatformIO Core installation is done again and again. Tried almost all fixes and solutions I could find in here and related bugs. Removed .atom and .platformio completely and tried installing from scratch. Same problem again. @ivankravets is there a way I could you provide more infos on my problem? I'm using Atom 1.17.0 on OSX 10.11.6 and tried to install platformio-ide 3.0.0-beta.3

KitPan, 2017-05-27:
I had the same issue exactly  as @saschaludwig OSX10.12.5 + 1.17.2, I dont even have brew or mac port installed, every solution found here were tried but none works, including clone from git and do npm install

ivankravets, 2017-05-27:
https://twitter.com/PlatformIO_Org/status/868394621851889664

ivankravets, 2017-05-27:
@KitPan I've just released hotfix for PIO Core 3.3.1. Please restart Atom. It should finish installation process. Please sorry.

KitPan, 2017-05-27:
@ivankravets Thanks! Really appreciated!