platformio/platformio-atom-ide #57: Auto-save on build

aperepel, 2016-02-28:
Hi folks, great effort on the IDE, looking forward to what it evolves into. Would like to request an option to save automatically on build. Tried out the IDE on a project and realized how annoying the prompt is on each build. As a long-time professional IDE user I expect it to save when I invoke the build action.

I'd prefer it even to be the default, but you're free adding this as a setting to the PlatformIO IDE.

orgkhnargh, 2016-02-28:
Go to "Settings > Packages > build" and check the "Automatically save on build" checkbox.

aperepel, 2016-02-28:
Cool, any way you can control it from the IDE module? I guess it's a different `build` module you are reusing, but locating those switches is non-trivial. Maybe add a doc note if nothing else?

ivankravets, 2016-03-06:
[image]

commit 92f7b564 (2016-03-06): Add settings for dependent packages to `Menu: PlatformIO > Settings` // Resolve #57
closed 2016-03-06 by ivankravets