platformio/platformio-atom-ide #58: Incorrect going project

alexey-su, 2016-02-29:
Create a new project for the Arduino Uno. Project name "servo". The project uses a LCD display. I create a file titled "servo.cpp". I connect h-files
```
#include <Wire.h> 
#include <LiquidCrystal_I2C.h>

void setup() {
	...
	lcd.print("PWM:");
	...
}
```
Everything is working.

Create a new project for the Arduino Uno. Project name "servoDrive". Copying "servo.cpp" file from the first draft. And change the label outputted
```
lcd.print("PWM2:");
```

The first mistake.
Clang error - include file "LiquidCrystal_I2C.h" not found.

The second mistake.
The project is translated correctly. Written to the controller. On the screen, instead of writing "PWM2:" I see the inscription "PWM:" from the previous recorded project "servo", rather than the current "servoDrive".

I rename file "servo.cpp" from project "servo Drive" to "servo Drive.cpp". clang parser error disappears. After filling in the controller firmware, I see the correct inscription "PWM2:".

ivankravets, 2016-02-29:
What do you mean under `Create a new project for the Arduino Uno. Project name "servoDrive".`?

Please use `Menu: PlatformIO > Initialize new PlatformIO project...` and copy ONLY `servo.cpp` to `src` directory from previous project

alexey-su, 2016-02-29:
Yes, and so it was done.
It created the first draft. Debugged. Everything is working.
Without leaving the atom, creating a second project. Create a new file with the same name as in the first draft. Opened two tabs. Copy-Paste. I change the text message displayed on the LCD screen.
An example of the LCD can be viewed on the network, for example, here http://cxem.net/arduino/arduino67.php

I create another project in the "sublime arduino stino". I moved the source code to the project. The project works.

Create a new project in PlatformIO Atom IDE. Created cpp file of the same name. moved the source code into a new file. Everything is working.

He returned to "servoDrive" project. Rename the file with "servo.cpp" on "servoDrive.cpp". Everything is working.

The impression that the system confuses file locations. After all, the atom open multiple files from different projects. I choose one of the open files in the project and I hope that the system recognizes the current project. For most platform PlatformIO no complaints. The project is error free. Which project gets to the controller - this is a big question.

ivankravets, 2016-02-29:
IDE uses project with the active opened file. If there no opened files, then it uses first added project in the tree.

I re-tested your behaviour and I've not found any problems. Please record some video or give us temporary sessions via TeamViewer or etc.

alexey-su, 2016-02-29:
In the evening check and I give the answer.

Can move into the Russian language?
You spoke in Russian on habrahabr.ru and geektimes.ru.

ivankravets, 2016-02-29:
@alexey-su Please send me PM to https://gitter.im/platformio/platformio

commit 4bd1bae1 (2016-03-07): Fix bug with incorrect determining of active project // Resolve #58
closed 2016-03-07 by ivankravets