platformio/platformio-atom-ide #59: Serial monitor keymap

Bladeslayer1000, 2016-03-03:
Hello,

I cant find the assignment for a keymap value to open the Serial monitor. Is there a option to open it with a keymap?

Greetings,
Bladeslayer1000

orgkhnargh, 2016-03-05:
Click "Edit > Open Your Keymap" ("Atom > Open Your Keymap" on Mac) and add the following to the file that opens:

```
'atom-workspace':
  'ctrl-shift-h': 'platformio-ide:maintenance.serial-monitor'
```

The `ctrl-shift-h` may be altered according to your needs. Put any (available) keystroke here. 

The `platformio-ide:maintenance.serial-monitor` is the name of the command that shows the Serial Monitor. You can find a list of available commands in here: https://github.com/platformio/platformio-atom-ide/blob/develop/menus/main.cson

This is also described in Atom docs: https://atom.io/docs/latest/using-atom-basic-customization#customizing-key-bindings

ivankravets, 2016-03-05:
@orgkhnargh need to add default keymap to our IDE. Please check which keystroke is available.

closed 2016-03-07 by ivankravets

Bladeslayer1000, 2016-03-13:
Thank you for the good support! Keep going.

ivankravets, 2016-03-13:
@Bladeslayer1000 don't forget to give us a star :star: https://github.com/platformio/platformio-atom-ide/stargazers

Bladeslayer1000, 2016-03-13:
Is it possible, to add the baudrate, and port automatically? So you go directly to the Serial monitor?

ivankravets, 2016-03-13:
Please rephrase your question.

commit 72f02fdd (2016-03-07): Add hotkey "cmd-shift-m / alt-shift-m" for Serial Monitor // Resolve #59