platformio/platformio-atom-ide #61: Doesn't work, can't install, download link buttons don't go to pages with a download.

teckel12, 2016-03-04:
Can't even start using this as the install steps are not clear and/or don't work.

orgkhnargh, 2016-03-04:
Are you saying that download links on [this page](http://platformio.org/#!/platformio-ide) do not work?

teckel12, 2016-03-04:
Yes.  Click the blue button.

teckel12, 2016-03-04:
Also, I believe Python is required, where's the link to that?

orgkhnargh, 2016-03-04:
You should probably use one of the green buttons if the page under the blue button makes no sense for you.

Link to the Python website appears during installation process if Python was not found on the user's computer.

We're sincerely sorry if anything on the website confuses you. Thank you for letting us know about all the confusing spots.

teckel12, 2016-03-04:
This website gives me heartburn and needs to be redone from the ground-up.  Classic example of people being too close to it to understand how to communicate to outsiders.  Here's what the instructions should be:

1) Install Python (be sure to click the "Add Python to PATH" box): https://www.python.org/downloads/
2) If you didn't click the "Add Python to PATH" option in step 1, uninstall Python and reinstall being sure to click the "Add Python to PATH" box).
3) Install PlatformIO: [link to install file]

You can mention that PlatformIO runs on Atom, but you don't need it as the download page implies as the PlatformIO install includes Atom!  Very confusion.  But still, even with the above instructions it still doesn't work.  When I run Atom I get a red error that says

"'platformio' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process.".

Well, when I installed Python I clicked the box to add it to my path.  And open an issue where?  No clue as to what the "package spawning the process" is to open an issue.

As I said, the download links don't work, are not clear, don't link to or even mention Python, and even once you jump through all the hoops it still doesn't work.

ivankravets, 2016-03-04:
Dear @teckel12,

We've just received your letter. Please sorry for the troubles. We will fix them as soon as possible and write here.

P.S: For you information, we 100% open source organization and propose users PlatformIO ecosystem totally for free. What is more, our site is open source too https://github.com/platformio/platformio-web . You make PR to it.

To be as non-profitable organization we are trying to improve all instruments from PlatformIO ecosystem, including Web site.

Please read this post https://github.com/platformio/platformio/issues/490

Regards, [PlatformIO Team](http://platformio.org/#!/contact).

teckel12, 2016-03-04:
To imply that it's built on top of Atom, that suggests that you need Atom for PlatformIO to function.  And there's even a button to install Atom, yet from that button you can't actually install Atom.  Once you abandon that, you find out that you don't really need to install Atom first, as implied.  Even then, PlatformIO still won't function the first time you run it.

teckel12, 2016-03-04:
I'm only here because someone reported an "issue" about one of my Arduino libraries, that it doesn't support PlatformIO.  Trying to figure out why this is an "issue" and what PlatformIO is led me here.  The learning cost of entry for PlatformIO is far too high.  I can't install it, I don't understand why I need it, nor why my Arduino library needs to be compatible with it.  It seems to me that the Arduino IDE works just fine for ATMega and ARM development.  Simple and one place for everything.  Further, I can install it with a single install and it works the first time.

It seems to me that if every library needs to be made compatible with PlatformIO to work with PlatformIO, the usefulness of PlatformIO is next to zero.  What am I thinking that's wrong or not understanding here?  Why must my Arduino library be PlatformIO compatible to be used?

ivankravets, 2016-03-04:
@teckel12 we have just updated our site:

* http://platformio.org/#!/platformio-ide
* http://platformio.org/#!/get-started

Does it look better?

Thanks a lot for the tips!

ivankravets, 2016-03-04:
>  I don't understand why I need it, nor why my Arduino library needs to be compatible with it.

PlatformIO is not replacement to Arduino IDE. PlatformIO is independent ecosystem starting from build system which was oriented to be run from CLI and finishing with powerful IDE and Library Manager.

The question is not related to "should I use/support PlatformIO?". We have here common user who use your library and our ecosystem. I propose to focus on it. Could you provide the info for the problems with your library and PlatformIO?

As for PlatformIO vs Arduino. We don't have common things. The only single thing is the source code of  `framework-arduino`. We have own cross platform build system, powerful library manager and other professional instruments. Someone likes to use NotePad to write code, others like to use professional IDE. However, we are working on the polishing our "products" which should be powerful inside but should  be simple for using as NotePad.

teckel12, 2016-03-04:
That's much more clear on the installation.  Still, it doesn't seem to work as even with a Python install and the path box checked, when I start Atom it gives me a PlatformIO error about the path as noted above.  There's something else setup, installed, or configured in your system that allows this to work, but not on a system that doesn't have any of the prerequisites (as on my system).

As far as the "issue" with my Arduino library--I have no idea what the problem is or even if there is one.  I got an an "issue" submitted that stated the following:

"It would be nice to have access to this library in PlatformIO manager:
http://docs.platformio.org/en/latest/librarymanager/creating.html"

Why wouldn't an Arduino library work unless the above is done?  How would adding this file to my distribution ZIP file make something work that's not working now.  I don't get it, and I don't get what is being asked or what PlatformIO is because I can't even run it to find out.

ivankravets, 2016-03-05:
> Why wouldn't an Arduino library work unless the above is done? 

In 99% all Arduino-based libs are compatible with @PlatformIO. Could you provide source code of your lib? We will test it.

Thanks.

teckel12, 2016-03-07:
Here's the library:

https://bitbucket.org/teckel12/arduino-new-ping/downloads

ivankravets, 2016-03-07:
@valeros please test it https://github.com/platformio/platformio-atom-ide/issues/61#issuecomment-193244837

valeros, 2016-03-07:
@teckel12 Your lib [compiled](https://gist.github.com/valeros/342ecb560717bdf698d8) without any errors.

teckel12, 2016-03-07:
The "issue" wasn't specifically that it wouldn't compile, it was that it wasn't compatible in some way.  He just replied back with this info:

> You just provide information where your library is located (bitbucket) + supported platforms in one manifest file (added in your repository). Then crawler will look up for your library and update information in library database based on this manifest.

His "issue" (I guess) is that he would need to install my library manually instead of automatically installing the dependencies through PlatformIO.  I'm just guessing, as I don't comprehend what PlatformIO is nor can I install it to find out.  But, I assume some kind of file would need to be created and I would need a repository of types.  Who knows, as I don't get any of this.

ivankravets, 2016-03-07:
@teckel12 You need to [create library.json](http://docs.platformio.org/en/latest/librarymanager/creating.html) for your project. You can take a look at multiple examples for exiting libs http://platformio.org/#!/lib (goto the one of lib and click on `library.json` tab).

If you can provide this info (name, keywords, descrption) for your library here, we will create this file.

teckel12, 2016-03-07:
I see.  That probably won't work as I don't have a repository source for my library, only a zip file to download.  I've had nothing but problems and headaches when using version control repositories.  Doing my own multi-system development over the cloud is far easier than cryptic version control repositories like git.  Instead, I do it a 100% automated way that doesn't cause user confusion with multiple versions of a library (I also never need to commit, update, branch or whatever).

But, here's my library download, just in case it's even possible without a repository:

https://bitbucket.org/teckel12/arduino-new-ping/downloads

ivankravets, 2016-03-07:
@teckel12 PlatformIO Library Crawler doesn't require source code which should be placed in the repository.

Could you provide this info:

* keywords
* description

We will create for you this manifest and place in our repo https://github.com/platformio/platformio-libmirror/tree/master/configs

Thanks.

teckel12, 2016-03-07:
Got it, the examples I looked at all had a repository.  I believe this is what you're looking for:

{
  "name": "NewPing",
  "keywords": "ultrasonic, ping, sonar, distance, SR04, SRF05, SRF06, DYP-ME007, JSN-SR04T, Parallax PING",
  "description": "Ultrasonic ping sensor library that works with most sensors, can use a single interface wire, doesn't hang, can run in the background, and is smaller and faster that all other ping libraries.",
  "authors":
  {
    "name": "Tim Eckel",
    "email": "tim@leethost.com",
    "url": "https://bitbucket.org/teckel12/arduino-new-ping/wiki/Home"
  },
  "examples": [
    "https://bitbucket.org/teckel12/arduino-new-ping/wiki/Home#!simple-newping-sketch",
    "https://bitbucket.org/teckel12/arduino-new-ping/wiki/Home#!single-pin-sketch",
    "https://bitbucket.org/teckel12/arduino-new-ping/wiki/Home#!event-timer-sketch",
    "https://bitbucket.org/teckel12/arduino-new-ping/wiki/Home#!15-sensors-sketch"
  ],
  "version": "1.7",
  "downloadUrl": "https://bitbucket.org/teckel12/arduino-new-ping/downloads/NewPing_v1.7.zip",
  "frameworks": "arduino",
  "platforms":
  [
    "atmelavr",
    "teensy"
  ]
}

Still, it does appear that this would need to be updated for each new release.  This doesn't happen often with NewPing, but a version 1.8 probably will be released in the next month.

Thanks!

valeros, 2016-03-08:
Added http://platformio.org/#!/lib/show/176/NewPing

closed 2016-03-08 by teckel12

ivankravets, 2016-03-08:
@teckel12 Tim, we will fix our site where will show Python Guide for Windows users and post here.

ivankravets, 2016-03-08:
@teckel12 We release new version of our IDE yesterday. Do you still have problems to try it?