platformio/platformio-atom-ide #62: Error with serial monitor (fix available)

periloso, 2016-03-05:
I'm trying to use a chinese ttl <> usb converter to program an arduino uno, and the serial monitor doesn't works by default.
The problem is that the official kext for the chipset creates a (very) weird device inside /dev that has spaces inside it (it's called /dev/cu.wch ch341 USB=>RS232 1470), so the serial monitor throws these errors:

```
Mac-Pro-di-Alessio:arduino_automac bigalex$ pio -f -c atom serialports monitor --port /dev/cu.wch ch341 USB=>RS232 1470 --baud 115200 --parity N --filter default --eol CRLF
Usage: pio serialports monitor [OPTIONS]

Error: Got unexpected extra arguments (ch341 USB= 1470)
Mac-Pro-di-Alessio:arduino_automac bigalex$
```

The problem can be easily fixed adding a " before and after the command string, like this:
```
      command += ` --${key} "${settings[key]}"`;
//      command += ` --${key} ${settings[key]}`;
```

Here's the fixed command.js:
[command-fixed.zip](https://github.com/platformio/platformio-atom-ide/files/159651/command-fixed.zip)

orgkhnargh, 2016-03-06:
Hi @periloso, thanks for pointing this out! I've recently pushed the fix. It would be nice if you test it with the latest IDE. You can find a guide on installing the development version of the IDE [here](https://github.com/platformio/platformio-atom-ide/issues/48#issuecomment-189379370).

periloso, 2016-03-06:
Well, it works, but the file i'm using (.atom/packages/platformio-ide/lib/serial-monitor/command.js) is still my own file, so i'm not that sure that switching to the development version did update the serial monitor :|

Besides that, since I need to view the binary data I'm sending in hex, I am using "CoolTerm" for mac that has this feature. After checking miniterm's source code, it seems that it can also show the data in hex... it would be a really cool feature to have in your IDE! :P

ivankravets, 2016-03-06:
@periloso please try http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#cmdoption-platformio-serialports-monitor--encoding . Does it work for you?

periloso, 2016-03-06:
@ivankravets, sadly nope: when I press ctrl+t and ctrl+a to change the encoding, i am asked the new encoding scheme; after writing hexlify the monitor dies throwing errors.
If instead I try to set it run with the --encoding "hexlify", it dies complaining for the same error.
Here's what happens:

```
--- Enter new encoding name [UTF-8]: ͢hexlify
Exception in thread tx:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 808, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 761, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/bigalex/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 432, in writer
    self.handle_menu_key(c)
  File "/Users/bigalex/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 526, in handle_menu_key
    codecs.lookup(new_encoding)
  File "/Users/bigalex/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 19, in <lambda>
    codecs.register(lambda c: hexlify_codec.getregentry() if c == 'hexlify' else None)
  File "/Users/bigalex/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/hexlify_codec.py", line 97, in getregentry
    _is_text_encoding=True,
    TypeError: __new__() got an unexpected keyword argument '_is_text_encoding'

    --- exit ---

    Mac-Pro-di-Alessio:arduino_automac bigalex$ pio -f -c atom serialports monitor --port "/dev/cu.wch ch341 USB=>RS232 1470" --baud "115200" --parity "N" --filter "default" --eol "CRLF" --encoding "h
    exlify"
    Error: __new__() got an unexpected keyword argument '_is_text_encoding'
```

Weird question but, is it implemented yet insite the code?
Thanks for the help! :)

ivankravets, 2016-03-06:
Try to run in system Terminal
```
pio -f -c atom serialports monitor --baud 115200 --encoding "hexlify"
```

If `pio` is not available, use Menu `PlatformIO: Install Shell Commands`

periloso, 2016-03-06:
Still the same error:
```
Mac-Pro-di-Alessio:arduino_automac bigalex$ pio -f -c atom serialports monitor --baud 115200 --encoding "hexlify"
Error: __new__() got an unexpected keyword argument '_is_text_encoding'
```
```pio``` was unavailable, but running on the ide terminal solved it :)

ivankravets, 2016-03-06:
@periloso please report this issue here https://github.com/pyserial/pyserial/issues

periloso, 2016-03-06:
Okay, I've [opened the bug there](https://github.com/platformio/platformio-atom-ide/issues/62#issuecomment-192988684); is there something I can help you with about the incorrect string passed. In case you're working on that part, it would be useful to have a field to pass some other options to miniterm inside the ide :)

ivankravets, 2016-03-06:
>  it would be useful to have a field to pass some other options to miniterm inside the ide :)

@orgkhnargh how about to add `INPUT` field where user be able to add extra flags/options manually?

orgkhnargh, 2016-03-06:
> it would be useful to have a field to pass some other options to miniterm inside the ide

This is unnecessary, since you can run the monitor in a terminal passing whatever option you need directly. Just open a new terminal (there is a "+" button in the status bar) and type `pio serialports monitor --whatever-option "value"`. Also you can open the Serial Monitor form in IDE, start a monitor, stop it (Ctrl+]) press ↑ (arrow up), add some options and run the modified command.

> the file i'm using (.atom/packages/platformio-ide/lib/serial-monitor/command.js) is still my own file

That's weird. IDE should have been deleted completely once you run `rm -r ~/.atom/packages/platformio-ide`. Are you sure you have executed the commands from [this message](https://github.com/platformio/platformio-atom-ide/issues/48#issuecomment-189379370)?

ivankravets, 2016-03-07:
> Okay, I've opened the bug there; 

@periloso , PlatformIO uses @pyserial [Miniterm](https://pythonhosted.org/pyserial/tools.html#miniterm)

Try 
```
python -m serial.tools.miniterm "/dev/cu.wch ch341 USB=>RS232 1470" 115200 --encoding "hexlify"
```

Please report this issue here https://github.com/pyserial/pyserial/issues

closed 2016-03-07 by ivankravets