platformio/platformio-atom-ide #63: Loop in atom updater

xxv, 2016-03-05:
Each time I start atom (for platformio), I get a window that pops up saying:

> PlatformIO IDE has been successfully installed!
> 
> However, some of its components will only become available after Atom window reload. You can click "Reload now" button below to perform reload immediately, or click "Reload later" and perform reload yourself with "View > Reload" command whenever you're ready.

with a "Reload now" and "Reload later" button. If I reload now, it restarts the IDE and then pops up again. If I press "Reload later" it goes away until I restart atom.

Versions: IDE 1.0.4 | CLI 2.8.4

Running on Debian stretch/sid

orgkhnargh, 2016-03-05:
This is probably a duplicate of #55. Check [this](https://github.com/platformio/platformio-atom-ide/issues/55#issuecomment-189369473) out. 
Basically, this is happening when there is a directory in your home directory, which is named as one of the IDE's dependencies. Please check if there is a directory in your home directory called as one of these: `build, autocomplete-clang, linter, linter-gcc, platformio-ide-terminal, language-ini, tool-bar`. If there is, rename it and see if the bug is gone. If the bug is gone, then do not worry -- it is already fixed and will be released soon.

xxv, 2016-03-06:
Yup. I have a `build` directory in my home directory. Thanks!

closed 2016-03-06 by xxv