platformio/platformio-atom-ide #64: "Installation has failed" error on windows 10

PhilippeBonnaz, 2016-03-07:
When trying to install platforio IDE with "platformio-atom-windows" installer I get the following error message: 
"Installation has failed
There was an error while installing the application. Check the setup log for more information and contact the author"

Here is the setup log file, the problem seems to be a too long path name, but I have no idea how to solve that issue

[PlatformioInstallErrorLog.txt](https://github.com/platformio/platformio-atom-ide/files/161610/PlatformioInstallErrorLog.txt)

ivankravets, 2016-03-07:
Do you have Atom already installed?

PhilippeBonnaz, 2016-03-07:
I get the exact same error message when trying to install Atom.

And I think the same Error log

[AtomInstallErrorLog.txt](https://github.com/platformio/platformio-atom-ide/files/161620/AtomInstallErrorLog.txt)

ivankravets, 2016-03-07:
Try to remove this folder `C:\Users\consultant.DAV-IDF-CON-PBO\AppData\Local\atom`

PhilippeBonnaz, 2016-03-07:
Still the same error

[AtomInstallErrorLog2.txt](https://github.com/platformio/platformio-atom-ide/files/161645/AtomInstallErrorLog2.txt)

ivankravets, 2016-03-07:
Could you try this archive (not installer) https://github.com/atom/atom/releases/download/v1.5.4/atom-windows.zip

Does Atom start?

cbentle, 2016-03-08:
I promise you that the issue is exactly the same issue I had. My solution was to remove the installer from the downloads folder, run it with administrator privileges even if you are already an administrator, you need to explicitly run its as administrator on Windows by right clicking. Once installed, platformio wouldn't spawn due to my Antivirus App firewall,  so I had to add python, atom, and all of the build tools to the white list. It is easiest to run the build command and then as each tool in the chain tries to run, allow it through your Antivirus pop up notification, I had to run build 8 times or so until it was all white listed.

PhilippeBonnaz, 2016-03-08:
Hi,
Yes atom is starting with that archive !

PhilippeBonnaz, 2016-03-08:
cbentle: I still get the exact same error message and error log, when trying to install atom or platforiom as administrator from outside the download folder ;)

ivankravets, 2016-03-08:
> Yes atom is starting with that archive !

Now install our package manually. See http://docs.platformio.org/en/latest/ide/atom.html#manual-installation

Step 2

PhilippeBonnaz, 2016-03-09:
Thank you !
Platformio installation worked

ivankravets, 2016-03-09:
@PhilippeBonnaz can you confirm that successfully built some project?

PhilippeBonnaz, 2016-03-09:
Yes Ivan, I'll let you know ;)

PhilippeBonnaz, 2016-03-09:
I hame having difficulties building the project.
No problem setting up the project (FYI the board I'm using is ST NUCLEO-L152RE)
I create the main.cpp file in src directory, copy the example project.
And get an "arduino.h" no such file or directory when building

[ArduinoNoSuchFile.txt](https://github.com/platformio/platformio-atom-ide/files/165326/ArduinoNoSuchFile.txt)

valeros, 2016-03-09:
Hi @PhilippeBonnaz please take a look at our [mbed Blink example](https://github.com/platformio/platformio/tree/develop/examples/mbed/mbed-blink)

PhilippeBonnaz, 2016-03-09:
Thank you @ivankravets and @valeros now it is working

closed 2016-03-10 by ivankravets