platformio/platformio-atom-ide #66: After automatic install I get error 'platformio' could not be spawned...

zakamik, 2016-03-13:
System: windows 8.1 x64
Python: 3.5.1 x64
Automatic install via binary from the site.
At every launch I get red popup window with message:
`'platformio' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process.`

checked PATH, found C:\Users\user\AppData\Local\atom\bin and also C:\Program Files\Python35\Scripts\;C:\Program Files\Python35\

ivankravets, 2016-03-13:
Please uninstall Python 3.  PlatformIO requires Python 2.7. See docs.

zakamik, 2016-03-13:
Confirmed, works after reinstall Python 2.7
However, allowing setup with incompatible environment is not an expected behavior of the automatic installer.

ivankravets, 2016-03-13:
@orgkhnargh when `'platformio' could not be spawned.`, please check Python version and explain user that Python 2.7 is required

orgkhnargh, 2016-03-14:
It is now not possible to install the IDE without Python 2 (as of d083f93ff7fae8d3269e5ebda7293c485217ce68). Both GUI is now more clear about the required Python version. Thanks for pointing to this!

closed 2016-03-14 by orgkhnargh