platformio/platformio-atom-ide #68: Uncaught Error: spawn platformio ENOENT

kuro114, 2016-03-15:
So this is a clean install and I have only Python 2.7 installed. The following error pops up on launch and I've poked around but to no avail. Any idea on how to make it go away? 

'platformio' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process.

Uncaught Error: spawn platformio ENOENT

orgkhnargh, 2016-03-15:
Did the installation process go fine? Have you tried to reinstall the IDE?

kuro114, 2016-03-15:
Yup restarted computer, uninstalled, reinstalled, and stuck. Not sure how to proceed.

orgkhnargh, 2016-03-15:
Which OS is it on?

kuro114, 2016-03-15:
WIndows 10

ivankravets, 2016-03-15:
Try to open system Terminal (cmd?) and paste this `python --version`. What do you see?

kuro114, 2016-03-15:
PS C:\Users\Spartan114> python --version
python : The term 'python' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the
name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ python --version
+ ~~~~~~
    + CategoryInfo          : ObjectNotFound: (python:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

ivankravets, 2016-03-15:
You have received the answer for your question. Please perform step 2 http://docs.platformio.org/en/latest/ide/atom.html#installation

ivankravets, 2016-03-23:
Please upgrade PlatformIO IDE to the latest 1.1.0 http://community.platformio.org/t/platformio-ide-1-1-0-and-platformio-cli-2-8-6/73/1

closed 2016-03-23 by ivankravets