platformio/platformio-atom-ide #71: UnicodeDecodeError: 'utf8' codec can't decode byte

steronydh, 2016-03-15:
On Windows 7, 64-bit

I'm getting this error with a newly-imported Arduino project:

UnicodeDecodeError: 'utf8' codec can't decode byte 0xef in position 119: invalid continuation byte:
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\main.py", line 151:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\...\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\...\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\scripts\atmelavr.py", line 121:
target_elf = env.BuildProgram()
File "C:\Users\...\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 55:
deplibs = env.BuildDependentLibraries("$PROJECTSRC_DIR")
File "C:\Users\...\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 347:
deplibs = _get_dep_libs(src_dir)
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 304:
state = _process_src_dir(state, env.subst(src_dir))
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 315:
state = _parse_includes(state, env.File(join(root, f)))
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 332:
_parse_includes(state, env.File(finder.getIncPath()))
File "C:\Users\...\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 320:
matches = INCLUDES_RE.findall(node.get_text_contents())
File "C:\Users\...\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 2684:
return contents[len(codecs.BOM_UTF8):].decode('utf-8')
File "c:\users\...\.atom\packages\platformio-ide\penv\lib\encodings\utf_8.py", line 16:
return codecs.utf_8_decode(input, errors, True)
 [ ERROR ] Took 0.73 seconds

Any suggestions welcome!

ivankravets, 2016-03-15:
Which is your user name?

See http://docs.platformio.org/en/latest/faq.html#windows-unicodedecodeerror-ascii-codec-can-t-decode-byte

steronydh, 2016-03-15:
I don't have non-latin characters in my username.

ivankravets, 2016-03-15:
Please provide
```
python --version
```

steronydh, 2016-03-15:
I have just tried the fix you provided by putting in "home_dir = C:\.platformio" and I get the same error!

I have Python 2.7.10

ivankravets, 2016-03-15:
Please try to uninstall Python and install the latest 2.7.11. Thanks.

steronydh, 2016-03-16:
I get the same error.

ivankravets, 2016-03-16:
@steronydh Could you provide TeamViewer session for debug?

steronydh, 2016-03-16:
Sorry no, problem is on work computer.

ivankravets, 2016-03-16:
Ok, let's resolve it remotely. Try to install PlatformIO CLI. Please forget about PlatformIO IDE for a while.
See http://docs.platformio.org/en/latest/installation.html
```
pip install platformio
platformio boards
```

Does it work?

steronydh, 2016-03-16:
Yes, that's working fine.

ivankravets, 2016-03-16:
Now
1. Open PlatformIO IDE
2. Goto `Menu: PlatformIO > Settings > platformio-ide`
3. Uncheck "Use built-in PlatformIO`
4. Restart IDE

steronydh, 2016-03-16:
That's done.

ivankravets, 2016-03-16:
Does it work now?

steronydh, 2016-03-16:
Same error unfortunately!

ivankravets, 2016-03-16:
Also with the same path `File "C:\Users....atom\packages\platformio-ide\penv\l` ?

steronydh, 2016-03-16:
No, it's:

UnicodeDecodeError: 'utf8' codec can't decode byte 0xef in position 119: invalid continuation byte:
File "C:\python27\lib\site-packages\platformio\builder\main.py", line 151:
env.SConscript("$BUILD_SCRIPT")
....

steronydh, 2016-03-16:
From the CLI, the full report is:

[03/16/16 09:52:04] Processing megaatmega2560 (platform: atmelavr, board: megaatmega2560, framework: arduino)
---------------------------------------------------------------------------------------------------------------
UnicodeDecodeError: 'utf8' codec can't decode byte 0xef in position 119: invalid continuation byte:
File "C:\python27\lib\site-packages\platformio\builder\main.py", line 151:
env.SConscript("$BUILD_SCRIPT")
File "C:\Temp\platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Temp\platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "C:\python27\lib\site-packages\platformio\builder\scripts\atmelavr.py", line 121:
target_elf = env.BuildProgram()
File "C:\Temp\platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 55:
deplibs = env.BuildDependentLibraries("$PROJECTSRC_DIR")
File "C:\Temp\platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 347:
deplibs = _get_dep_libs(src_dir)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 304:
state = _process_src_dir(state, env.subst(src_dir))
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 315:
state = _parse_includes(state, env.File(join(root, f)))
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 332:
_parse_includes(state, env.File(finder.getIncPath()))
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 332:
_parse_includes(state, env.File(finder.getIncPath()))
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 320:
matches = INCLUDES_RE.findall(node.get_text_contents())
File "C:\Temp\platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 2684:
return contents[len(codecs.BOM_UTF8):].decode('utf-8')
File "c:\python27\lib\encodings\utf_8.py", line 16:
return codecs.utf_8_decode(input, errors, True)
========================================= [ ERROR ] Took 0.67 seconds =========================================

ivankravets, 2016-03-16:
`$PROJECTSRC_DIR`, `_parse_includes`: are you sure that these paths don'r contains problematic chars?

Please create project in `C:\myproject` ( http://docs.platformio.org/en/latest/quickstart.html#initialize-project ). Place here minimal code. Change `platformio.ini` to http://docs.platformio.org/en/latest/faq.html#windows-unicodedecodeerror-ascii-codec-can-t-decode-byte

steronydh, 2016-03-16:
I can get that to compile! :)

ivankravets, 2016-03-16:
Ok, are you sure that you don't have 'non-unicode` chars in the path? Which is the name of your user?

steronydh, 2016-03-16:
How would I recognise a "non unicode" char? It's Dev23.

ivankravets, 2016-03-16:
Ok, let's try to comment in `platformio.ini` the 
```ini
; [platformio]
; home_dir
```

steronydh, 2016-03-16:
No luck. However, I **can** compile the simple test in my usual Arduino folder, so the usual path is OK.

steronydh, 2016-03-16:
I've just tried adding a library to the simple test program, and it's compiled up until this error:

scons: *** [.pioenvs\megaatmega2560\MyLib\MyCalibration.o] UnicodeDecodeError : 'utf8' codec can't decode byte 0xef  in position 119: invalid continuation byte

ivankravets, 2016-03-16:
Could you specify FULL PATH to `.pioenvs` dir? This dir is created in your project directory.

steronydh, 2016-03-16:
C:\Users\Dev23\Documents\Arduino\testio\.pioenvs

ivankravets, 2016-03-16:
That is very interesting... Please provide me an output of
```
python -c "from platformio import util; print util.get_home_dir()"
```

steronydh, 2016-03-16:
Depending on whether or not I have home_dir specified, I get either:

C:\myproject> python -c "from platformio import util; print util.get_home_dir()"
C:\Temp\platformio

or

C:\myproject> python -c "from platformio import util; print util.get_home_dir()"
C:\Users\Dev23\.platformio

The build error occurs either way.

ivankravets, 2016-03-16:
Could you share archived project somewhere? 
`matches = INCLUDES_RE.findall(node.get_text_contents())` means that in some of your project file is non-unicode chars (bad saved file).

Please try in IDE generic "Blink" example http://docs.platformio.org/en/latest/ide/atom.html#quick-start

Does it work for you?

steronydh, 2016-03-16:
Ah, I think that's the problem. I do have some KOI8-R text (Russian / Cyrillic) in one .h file in a library, could that cause the problem? I'm not using that specific file in the sketch though.

All of the file names are in Unicode/English. The file isn't corrupt though, it was intentionally saved like that!

Sorry for taking up so much of your time today, thank you for your great support.

ivankravets, 2016-03-16:
1. Please uninstall PlatformIO which is installed globally. Open system terminal and `pip uninstall platformio`
2. Go to `Menu: PlatformIO > Settings > platformio-ide` and enable again `Use builtin PlatformIO`
3. Copy the content of your file which contains KOI8-R, delete it. Go to PlatformIO IDE, create new file, paste content. Save. It should work.

The problem is not linked with PlatformIO. As you can see, the file is not saved as UTF-8 and Python can't read it.

steronydh, 2016-03-17:
I found a simple workaround: I renamed the header file from ru_RU.h to ru_RU.h.txt and platformio ignored it. I had to add some #includes to the .ino file but it's working now! It's even fixed the slow compile time of Arduino IDE 1.6.8. Great!

Thanks again for all your help, Ivan.

ivankravets, 2016-03-17:
It works in Arduino IDE because it doesn't parse user's C/C++ related files. In PlatformIO, we scan user's project and find all dependencies. The service is called as PlatformIO Library Dependency Finder (LDF). It lives in our Build System.

LDF helps developers to don't bother which library management. If it see that some lib is used, it will automatically append it to build process, build to static lib and include to final firmware.

------

Please send me@ikravets.com the archive with "ru_RU.h" and I will try to understand why Python can't load it. This is very important for us. Thanks.

steronydh, 2016-03-17:
To open CSV files containing KOI8-R text I've had to use this Python to read the Russian properly:

`import csv, codecs`
` `
`with codecs.open('koi8-r.csv','r', 'koi8-r') as ip:`
`...`

ivankravets, 2016-03-17:
Why you don't keep files in UTF-8?

steronydh, 2016-03-17:
The library we're using on Arduino to draw text on an OLED display doesn't support UTF8.

ivankravets, 2016-03-17:
Thanks Steve. Please don't hesitate to contact us if will have other questions/problems.

closed 2016-03-17 by ivankravets