platformio/platformio-atom-ide #74: Avrdude progress not showing

steronydh, 2016-03-18:
This might be due to the way the platformio terminal captures avrdude's output, but when programming the flash, the avrdude progress bar doesn't appear until the end, when it appears at 100%. It would be useful with larger projects to see the progress during programming.

ivankravets, 2016-03-23:
Please upgrade to the latest PlatformIO IDE 1.1.0. Could you reproduce this issue?

steronydh, 2016-03-23:
Two of my libraries are unable to find Arduino.h:

ivankravets, 2016-03-23:
Please run in PlatformIO IDE Terminal
```
pio platforms uninstall atmelavr
```

Now, try to build again

steronydh, 2016-03-23:
There is an error:

Uninstalling framework-arduinoavr package:      Error: Traceback (most recent call last):
  File "c:\users\steve\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\__main__.py", line 107, in main
    cli(None, None, None)
  File "c:\users\steve\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\steve\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
....
  File "C:\Python27\Lib\shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "C:\Python27\Lib\shutil.py", line 254, in rmtree
    os.rmdir(path)
WindowsError: [Error 145] The directory is not empty: 'C:\\Users\\Steve\\.platformio\\packages\\framework-arduinoavr'

ivankravets, 2016-03-23:
Please delete manually `C:\Users\Steve\.platformio` directory

steronydh, 2016-03-24:
I'm seeing:

avrdude: Device signature = 0x1e9801
avrdude: reading input file ".pioenvs\megaatmega2560\firmware.hex"
 
avrdude: writing flash (85376 bytes):
 
Then nothing until the write is complete.

ivankravets, 2016-03-24:
Could you try `Menu: PlatformIO > Run other target...` and select `Program` instead upload

steronydh, 2016-03-24:
I don't see an option `Program` but if I select `PlatformIO > Run other target > Upload using programmer` I get an error:

 avrdude: ser_open(): can't open device "\\.\com1": The system cannot find the file specified.
Using Port                    : com1
Using Programmer              : wiring

ivankravets, 2016-03-24:
See docs http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer. Please add these lines to your `platformio.ini`
```ini
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_port = INSERT_PORT_HERE
```

You need to replace `INSERT_PORT_HERE` with the real serial port. Use `Menu: PlatformIO > List Serial Ports`. For example,
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_port = /dev/cu.usbmodemFD1311
```

Now, use `Uploading using Programmer`.

P.S: We do it only for testing to understand the problem. Thanks.

steronydh, 2016-03-24:
I now get this:

avrdude: AVR device initialized and ready to accept instructions
 
Reading | ################################################## | 100% 0.04s
 
To disable this feature, specify the -D option.
avrdude: Device signature = 0x1e9801

avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
avrdude: erasing chip
avrdude: stk500v2_command(): command failed
 
 
avrdude done.  Thank you.
scons: *** [program] Error 255

ivankravets, 2016-03-24:
Hm...
1. Please remove the latest added lines from `platformio.ini`: `upload_flags` and `upload_port`.
2. We will not use `Upload using Programmer`
3. Do you have Arduino IDE? Please try to flash your board with it. Does it work?

steronydh, 2016-03-24:
It flashes fine with the Arduino IDE and Platformio Upload, just not Upload using Programmer. :)

steronydh, 2016-03-24:
This isn't a problem with small sketches, it's only really with big ones that it's useful to see the progress! The Arduino IDE shows the flashing progress as a line of ### when you turn on verbose output.

ivankravets, 2016-03-24:
Try to open `PlatformIO > Terminal` and paste directly `pio run -t upload`. Does it work?

steronydh, 2016-03-24:
That uploads fine.

ivankravets, 2016-03-24:
You write that it uploads well via `Upload` button. The single problem is that you don't see progress? But board flashes well?

steronydh, 2016-03-25:
Yes, the board flashes. I think I haven't explained very well! Avrdude shows a row of # signs moving right as it flashes the board when you use the Arduino IDE. With the platformio terminal, it says that it's flashing, but you don't see the row of # signs until 100% when avrdude finishes.

ivankravets, 2016-03-26:
Do you see progress using PlatformIO IDE Terminal and `pio run -t upload`?

steronydh, 2016-03-29:
No, I get:

avrdude: writing flash (85840 bytes):

Then the next line appears all at once when the programming is finished:

Writing | ################################################## | 100% 21.49s

ivankravets, 2016-03-31:
@steronydh I've just took a look at code and found where this issue is hidden: https://github.com/platformio/platformio/blob/develop/platformio/util.py#L55

However, I don't see here compromise solution. If I output RAW data from subprocess, the final picture for the user will look awful. I apply colour scheme per line depending on `returncode` from the Async Communication. 

Nevertheless, this issue is related to PlatformIO CLI. Please create new here https://github.com/platformio/platformio. Maybe, some time I will think normal solution for that.

closed 2016-03-31 by ivankravets