platformio/platformio-atom-ide #75: Uncaught Error: Failed to get boards

mettinoktay, 2016-03-20:
As soon as I clicked the Initialize new PlatformIO Project or update existing button, this error was thrown.

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.6.0
**System**: Microsoft Windows 10 Home Single Language
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.0.5

### Stack Trace

Uncaught Error: Failed to get boards

```
At /C:/Users/Metin%20Oktay/.atom/packages/platformio-ide/lib/utils.js:206

Error: Failed to get boards
    at getBoards (C:/Users/Metin Oktay/.atom/packages/platformio-ide/lib/utils.js:163:13)
    at InitializeNewProjectView.initializeBoardsSelect (C:/Users/Metin Oktay/.atom/packages/platformio-ide/lib/init/view.js:67:20)
    at new InitializeNewProjectView (C:/Users/Metin Oktay/.atom/packages/platformio-ide/lib/init/view.js:63:10)
    at command (C:/Users/Metin Oktay/.atom/packages/platformio-ide/lib/init/command.js:30:14)
    at atom-workspace.platformioIdeInitializeNewProject (C:/Users/Metin Oktay/.atom/packages/platformio-ide/lib/main.js:60:54)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\Metin Oktay\AppData\Local\atom\app-1.6.0\resources\app.asar\src\command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (C:\Users\Metin Oktay\AppData\Local\atom\app-1.6.0\resources\app.asar\src\command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\Metin Oktay\AppData\Local\atom\app-1.6.0\resources\app.asar\src\command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (C:\Users\Metin Oktay\AppData\Local\atom\app-1.6.0\resources\app.asar\src\atom-environment.js:1013:28)
    at emitOne (events.js:77:13)
    at EventEmitter.emit (events.js:169:7)
```

### Commands

```
  2x -0:19.8.0 platformio-ide:initialize-new-project (atom-text-editor.editor.is-focused)
```

### Config

```json
{}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.8.9
build, v0.57.0
language-ini, v1.14.0
linter, v1.11.3
linter-gcc, v0.6.8
platformio-ide, v1.0.5
platformio-ide-terminal, v2.0.3
tool-bar, v0.4.0

# Dev
No dev packages
```

orgkhnargh, 2016-03-20:
Please, open a terminal (white `+` on the status bar), type `pio boards` and hit the Enter. Do you see any errors there?

mettinoktay, 2016-03-20:
Hey there. I ran the command under `C:\Users\username>` and here is the result:

`pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ pio boards
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException`

orgkhnargh, 2016-03-20:
Is the `Use built-in PlatformIO` checbox enabled? Click `PlatformIO > Settings > PlatformIO IDE` to see the settings.

orgkhnargh, 2016-03-20:
The issue, most probably, is due to the space in your username. It is a known problem. 

As a workaround you may set the `PLATFORMIO_HOME_DIR` environment variable to something without space (e. g., `C:\.platformio`) and restart the IDE. Here is a guide on how to edit an environment variables in Windows 10: http://superuser.com/a/949577

mettinoktay, 2016-03-20:
In the Environment Variables, there wasn't `PLATFORMIO_HOME_DIR`. So I first created `C:\.platformio` folder, then created `PLATFORMIO_HOME_DIR` environment variable and set the path to `C:\.platformio`. Yet same error is still being thrown.

orgkhnargh, 2016-03-20:
Click `View > Developer > Toggle Developer Tools`. New pane will open. Go to the `Console` tab of that pane (if the tab is not there, click on `>>` symbol on top of the pane to see a list of hidden tabs). There is a text input field on the bottom of the Console tab (it begins with `>` symbol). Type `process.env.PLATFORMIO_HOME_DIR` into that field and hit Enter. Does it print `C:\.platformio` back?

mettinoktay, 2016-03-20:
Thank you for your effort on solving the problem. Mysteriously, uninstalling PlatformIO IDE within Atom then re-installing it solved the problem.

orgkhnargh, 2016-03-20:
Great! Enjoy! :)

ivankravets, 2016-03-20:
@mettinoktay did you left `PLATFORMIO_HOME_DIR`?

mettinoktay, 2016-03-20:
@orgkhnargh Typing `process.env.PLATFORMIO_HOME_DIR` in Console returned `C:\.platformio`. 
@ivankravets I hope this also answers your question because I'm not sure if I know what you mean.

ivankravets, 2016-03-20:
@mettinoktay I would ask you to provide my all respective information according this issue. I know about this problem for a long time. PlatformIO doesn't work if user's name is not match with [a-z0-9] and I'm ready to deep in PlatformIO Core and fix it. Your USERPROFILE contains `space`. It's OK, don't change it. As I've said before, it's our problem.

1. Please remove  `PLATFORMIO_HOME_DIR` from system environment variables. 
2. Try to uninstall IDE, remove `C:/Users/Metin Oktay/.atom`
3. Install IDE again.

Do you have problems? Because in this case `C:/Users/Metin Oktay/.platformio` should be used.

mettinoktay, 2016-03-20:
@ivankravets @orgkhnargh Ivan, I've done as you instructed, and it works like charm. I don't know what's different in this latest installment of mine. Anyway, thank you very much!

ivankravets, 2016-03-20:
@mettinoktay Try to delete `C:\.platformio`. Then build something. Does IDE create again this `C:\.platformio`?

mettinoktay, 2016-03-22:
@ivankravets oh yeah it did.
[image]

ivankravets, 2016-03-22:
@mettinoktay thanks!

Did you move toolbar to the top?

mettinoktay, 2016-03-22:
@ivankravets It was at the top by default, I did nothing :-)

ivankravets, 2016-03-23:
Thanks! :) Looks like @orgkhnargh fixed it in https://github.com/platformio/platformio-atom-ide/commit/7ccaa89d3679d79a2e5eb1c6f532880704bbc232

closed 2016-03-23 by ivankravets

mavenius, 2016-04-01:
Just as an additional note, I had the same behavior, but my problem was that I am apparently not very good at following directions. Installing the correct version of Python (2.7 instead of 3) and reinstalling the IDE fixed it.