platformio/platformio-atom-ide #76: Error "'platformio' could not be spawned" after installing "PlatformIO IDE for Atom bundle"

innesm, 2016-03-22:
I chose to install PlatformIO IDE using the bundle because, reading the installation instructions, this looks like the recommended approach.

I am assuming that all the other installation steps are not needed if you go this route.

Screenshot:

[image]

innesm, 2016-03-22:
I uninstalled Atom then reinstalled from the plain atom installer, then noticed the same problem so uninstalled Atom and deleted the .atom folder and Atom folders below AppData, then reinstalled.

Then I installed the platformio-ide package in Atom. I got the same error again, so it looks like the problem isn't specific to the "bundle"

orgkhnargh, 2016-03-22:
Is there any non-ASCII characters in your username?

innesm, 2016-03-22:
No.

innesm, 2016-03-22:
I now see that this (install python 2.7) is a step to be done before installing the bundle. :dizzy_face:

orgkhnargh, 2016-03-22:
Did installing of Python 2.7 fix the issue?

innesm, 2016-03-23:
I tested changing my path so that python2.7 was first, but I still got the same error when running platformio-ide. In fact, running "procmon" showed no python-related activity when running atom.

ivankravets, 2016-03-23:
@innesm do you use the latest 1.1.0 version? Please `Menu: PlatformIO > Settings > platofrmio-ide` and update to the 1.1.0

innesm, 2016-03-23:
@ivankravets that fixed it - thanks! (I had installed from here http://platformio.org/platformio-ide)

ivankravets, 2016-03-23:
Thanks a lot for the report! Happy Coding!

Please give a star :star: for our repositories:
- https://github.com/platformio/platformio-atom-ide/stargazers
- https://github.com/platformio/platformio/stargazers

Thanks!

closed 2016-03-23 by ivankravets