platformio/platformio-atom-ide #77: "Arduino.h: No such file or directory" error after upgrading to 1.1.0

mettinoktay, 2016-03-23:
After updating to 1.1.0, PlatformIO has begun to throwing this error. 
[image]

ivankravets, 2016-03-23:
Please run in PlatformIO Terminal:
```
pio platforms list
```

And then delete all platforms
```
pio platforms uninstall platform1 paltformN
```

mettinoktay, 2016-03-23:
I received `Error: Unknown platform 'platform1'`.

ivankravets, 2016-03-23:
Of course you will receive it. It was an example. Please run `pio platforms list` and then delete all platforms listed in this command. For example (real), `pio platforms uninstall atmelavr`. Then try to build project.

mettinoktay, 2016-03-23:
Hahaha sorry, my mistake :smile: I ran `pio platforms uninstall atmelavr`. After I hit compile, PlatformIO redownloaded atmelavr. Now it works properly. Thanks!

ivankravets, 2016-03-23:
Happy Coding! :smile: 

Please give a star :star: for our repositories:
- https://github.com/platformio/platformio-atom-ide/stargazers
- https://github.com/platformio/platformio/stargazers

Thanks!

closed 2016-03-23 by ivankravets