platformio/platformio-atom-ide #78: Uncaught TypeError: Cannot read property 'toLowerCase' of undefined

brbrr, 2016-03-24:
[Enter steps to reproduce below:]

1. update to dev version
2. restart atom

**Atom Version**: 1.6.0
**System**: Mac OS X 10.11.3
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.1.0

### Stack Trace

Uncaught TypeError: Cannot read property 'toLowerCase' of undefined

```
At /Applications/Atom.app/Contents/Resources/app.asar/node_modules/pathwatcher/lib/directory.js:47

TypeError: Cannot read property 'toLowerCase' of undefined
    at new Directory (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/pathwatcher/lib/directory.js:47:39)
    at /Users/brbrr/.atom/packages/platformio-ide/lib/maintenance.js:268:54
    at Array.map (native)
    at Object.handleLibChanges (/Users/brbrr/.atom/packages/platformio-ide/lib/maintenance.js:226:16)
    at /Users/brbrr/.atom/packages/platformio-ide/lib/main.js:203:19
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:125:28)
    at Project.module.exports.Project.addPath (/Applications/Atom.app/Contents/Resources/app.asar/src/project.js:255:29)
    at AtomEnvironment.module.exports.AtomEnvironment.openLocations (/Applications/Atom.app/Contents/Resources/app.asar/src/atom-environment.js:1029:26)
    at EventEmitter.outerCallback (/Applications/Atom.app/Contents/Resources/app.asar/src/application-delegate.js:244:18)
    at emitTwo (events.js:92:20)
    at EventEmitter.emit (events.js:172:7)
```

### Commands

```
     -0:09.6.0 project-manager:list-projects (atom-text-editor.editor.is-focused)
  5x -0:08.6.0 core:move-down (atom-text-editor.editor.mini.is-focused)
     -0:07.3.0 core:confirm (atom-text-editor.editor.mini.is-focused)
```

### Config

```json
{
  "core": {
    "disabledPackages": [
      "quick-query",
      "data-atom",
      "linter-erb",
      "color-picker",
      "symbols-view",
      "gitignore-snippets",
      "project-plus",
      "terminal-plus",
      "imdone-atom",
      "build",
      "autocomplete-clang",
      "dash",
      "markdown-writer",
      "merge-conflicts",
      "todo-show",
      "api-docs"
    ],
    "ignoredNames": [
      ".idea",
      ".git",
      "*.lock",
      "*tags*"
    ],
    "themes": [
      "atom-dark-ui",
      "one-dark-syntax"
    ]
  },
  "platformio-ide": {
    "customPATH": "/Users/brbrr/.nvm/versions/node/v5.1.0/bin:/usr/local/bin:/Users/brbrr/.rbenv/shims:/usr/local/bin:/Users/brbrr/.rbenv/bin:/usr/lo\ncal/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/brbrr/Documents/Development/IdeaProjects/Selenium_Drivers:/opt/X11/bin:/usr/local/Mac\nGPG2/bin:/Users/brbrr/.atom/packages/platformio-ide/penv/bin:/usr/local/smlnj/bin:/usr/local/share/npm/bin/:/Users/brbrr/Developme\nnt/Utils/android-sdk-macosx/build-tools:/Users/brbrr/Development/Utils/android-sdk-macosx/platform-tools:/Users/brbrr/Development/\nUtils/android-sdk-macosx/tools",
    "useDevelopPlatformIO": true
  }
}
```

### Installed Packages

```coffee
# User
atom-ctags, v4.5.0
autoclose-html, v0.23.0
emmet, v2.4.3
language-arduino, v0.4.1
language-ini, v1.14.0
language-rspec, v0.5.0
linter, v1.11.4
linter-gcc, v0.6.8
linter-rubocop, v0.4.7
linter-ruby, v1.2.2
minimap, v4.21.0
minimap-find-and-replace, v4.5.0
minimap-git-diff, v4.2.0
platformio-ide, v1.1.0
platformio-ide-terminal, v2.0.3
project-manager, v2.7.6
ruby-block, v0.3.5
synced-sidebar, v0.4.3
tool-bar, v0.4.0
tool-bar-main, v0.0.9

# Dev
No dev packages
```

orgkhnargh, 2016-03-24:
Please run the following in a terminal: `python -c "from platformio import util; print util.get_lib_dir()"` from the project directory that was open when TypeError happened.

brbrr, 2016-03-24:
```
brbrr@revairx: weather_cli (master)*$ python -c "from platformio import util; print util.get_lib_dir()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named platformio
brbrr@revairx: weather_cli (master)*$ which platformio
/usr/local/bin/platformio
```

orgkhnargh, 2016-03-24:
Did you use your system terminal or a one from Atom (`+` button on the status bar)?

brbrr, 2016-03-24:
Was tested in both system and `pio-terminal` - same result mentioned before

orgkhnargh, 2016-03-24:
Is `Use built-in PlatformIO` option enabled? It's located under `PlatofrmIO > Settings > PlatformIO IDE`.

brbrr, 2016-03-24:
Yes, it's enabled

orgkhnargh, 2016-03-24:
Your `customPATH` contains a lot of stuff before `/Users/brbrr/.atom/packages/platformio-ide/penv/bin`. There must be other python found earlier. Can you clear `customPATH` temporary?

orgkhnargh, 2016-04-18:
Hi @brbrr,

Is this still an issue for you after the 1.1.2 release?

brbrr, 2016-04-18:
actually that error appear only once. so I suppose it's fixed.

orgkhnargh, 2016-04-18:
Great!

closed 2016-04-18 by orgkhnargh