platformio/platformio-atom-ide #79: Failed to run command: Build/Compile/Clean

brbrr, 2016-03-24:
Unable to use side bar/menu item for build/compile/clean commands. I got `Failed to run command: Build`

running it from terminal like `pio run` works fine. 

Using latest dev package on Mac OS
```
$ platformio upgrade
You're up-to-date!
PlatformIO 2.8.6 is currently the newest version available.
```

orgkhnargh, 2016-03-24:
Does it work when you use `PlatformIO > Run other target...`?

brbrr, 2016-03-24:
`Run other target`/`Update targets`/`Toggle build panel` don't have any visual effect.
I was worked for me few weeks ago(with previous package version(s))

orgkhnargh, 2016-03-24:
The config you've posted at #72 indicates that the `build` package is disabled. Please enable it back and try again.

brbrr, 2016-03-24:
great! thanks. Should pio inform users about uninstalled/disabled requirements?

orgkhnargh, 2016-03-24:
It does install and enable all requirements on start.

brbrr, 2016-03-24:
looks like it's not. I've enable pio package, and `build` was disabled. Also I switch to dev build, and `build` still wasn't enabled. 

in general - it would be nice to see more explaining tooltip for failed build command

orgkhnargh, 2016-03-24:
You're right. It does not activate dependencies, unless it also installed or upgraded anything. This is certainly a bug. Thanks for pointing this out.

> it would be nice to see more explaining tooltip for failed build command

Fair point. I will add more information to a tooltip.

orgkhnargh, 2016-04-19:
Should work fine now.

commit d962b851 (2016-04-19): Mention the `build` package when fails to run a target

Resolve #79
closed 2016-04-19 by orgkhnargh