platformio/platformio-atom-ide #80: Uncaught Error: ENOENT: no such file or directory, scandir 'atom://config'

punitsoni, 2016-03-25:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.6.0
**System**: Mac OS X 10.11.1
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.1.0

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, scandir 'atom://config'

```
At fs.js:813

Error: ENOENT: no such file or directory, scandir 'atom://config'
    at Error (native)
    at Object.fs.readdirSync (fs.js:813:18)
    at Object.fs.readdirSync (ATOM_SHELL_ASAR.js:454:28)
    at Object.ensureProjectsInited (/Users/punits/.atom/packages/platformio-ide/lib/maintenance.js:53:29)
    at /Users/punits/.atom/packages/platformio-ide/lib/main.js:202:19
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:125:28)
    at Project.module.exports.Project.addPath (/Applications/Atom.app/Contents/Resources/app.asar/src/project.js:255:29)
    at AtomEnvironment.module.exports.AtomEnvironment.openLocations (/Applications/Atom.app/Contents/Resources/app.asar/src/atom-environment.js:1033:26)
    at EventEmitter.outerCallback (/Applications/Atom.app/Contents/Resources/app.asar/src/application-delegate.js:244:18)
    at emitTwo (events.js:92:20)
    at EventEmitter.emit (events.js:172:7)
```

### Commands

```
```

### Config

```json
{
  "core": {
    "audioBeep": false,
    "excludeVcsIgnoredPaths": false
  }
}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.8.9
build, v0.57.0
language-ini, v1.14.0
linter, v1.11.4
linter-gcc, v0.6.8
platformio-ide, v1.1.0
platformio-ide-terminal, v2.0.3
tool-bar, v0.4.0

# Dev
No dev packages
```

ivankravets, 2016-03-26:
Please fill
```
[Enter steps to reproduce below:]

1. ...
2. ...
```

ivankravets, 2016-03-26:
I've just reproduced this bug:

1. Open empty Atom Window
2. Click on Setting (gear icon) in Toolbar

commit 1359582a (2016-03-26): Check project dir existence during ensureProjectsInited()

Fixes #80
closed 2016-03-26 by orgkhnargh