platformio/platformio-atom-ide #81: Arduino library

blasterbug, 2016-03-25:
Is there a way to automatically import libraries I use/develop with the Arduino IDE ?

ivankravets, 2016-03-25:
Could you explain what do you mean? If you install libs using Arduino IDE and later switched project to its libs, you need perform `Menu: PlatformIO > Rebuild C/C++ Index`.

blasterbug, 2016-03-25:
I mean, is there a way to use Arduino libraries, the ones I wrote just 
I used to do with the Arduino IDE ?

ivankravets, 2016-03-26:
@blasterbug Sure, just place your library to the `lib` directory in the project

blasterbug, 2016-03-26:
So I need to copy/paste the libs within the projects I need it... Not 
really efficient!

ivankravets, 2016-03-26:
In this case you need to place library to global `LIB` dir. See http://docs.platformio.org/en/latest/projectconf.html#lib-dir

ivankravets, 2016-06-23:
Resolved via `Menu: Import Arduino IDE Project...`

closed 2016-06-23 by ivankravets