platformio/platformio-atom-ide #82: Editor and gcc/lint panel scrolling bug

Deekers, 2016-03-25:
Attached are two screen shots: The first shows the gcc/lint bubble in the editor panel.

[image]

As the editor panel is scrolled the same bubble moves up and down with the scroll action and moves over top of the gcc/lint panel.

[image]

The fact that the gcc/lint bubble is actually pointing to an incorrect line is a separate problem.

Atom 1.6.0
OSX 10.11.2

orgkhnargh, 2016-03-25:
A bubble also covers a tab bar when you scroll down.

@Deekers Thank you for the report!

ivankravets, 2016-03-26:
Please forward this issue to Atom Linter package https://github.com/steelbrain/linter/issues

Thanks!

closed 2016-03-26 by ivankravets