platformio/platformio-atom-ide #89: Update .clang-complete and .gcc-flags.json with libraries locally added and scanned

abhishek-kakkar, 2016-04-02:
I've observed that PlatformIO is capable of detecting and compiling in libraries from the local lib/ folder in a project. However to obtain autocomplete I have to manually insert the lib path in the .clang-complete file.

Again the linter displays the error that it cannot find the header file corresponding to my user-defined library include path (the file compiles fine and everything works good though), it's a cosmetic issue.

It'd be nice to have the IDE detect the libraries automatically and update the .gcc-flags.json and .clang-complete files on rebuild to include the user-defined libraries during autocomplete features.

orgkhnargh, 2016-04-02:
The IDE is supposed to do that, unless you disabled the "Automatically rebuild C/C++ Project Index" setting. Can you please check your settings?

abhishek-kakkar, 2016-04-02:
One of the folders was a symlink to an existing library folder and it wasn't detecting it. I added another library to the folder (copied it into the lib folder) and it seems to be working now.

orgkhnargh, 2016-04-02:
How exactly did a symlink get into the `lib` directory? Was the IDE open when you put a symlink there?

It works fine for me. I did create a symlink with "ln -s /path/to/library /path/to/project/lib/" and the IDE updated both .gcc-flags.json and .clang-complete.

P. S. 
There is the "Rebuild C/C++ Project Index (Autocomplete, Linter)" command under the PlatformIO menu. You may find it more reliable than putting libraries paths into .gcc-flags.json and .clang-complete by hands.

abhishek-kakkar, 2016-04-02:
Yup, working fine now. I get notifications when I add something onto the library folder saying that the index is going to be rebuilt and it's working fine now. Thanks!

closed 2016-04-02 by abhishek-kakkar