platformio/platformio-atom-ide #92: "Error matching failed!" Notification after Build

jakematlick, 2016-04-10:
After I build a project (the build is successful), I get the following error notification.

> Error matching failed! / Error parsing regex. XRegExp is not a function
> http://imgur.com/1fjRAWr

I tested both an imported Arduino project (.ino) and a new project created with platformio (.cpp).
I tested two build targets (Arduino Uno and Teensy 3.0).

I am running platformio IDE in Atom on Mac OS X 10.11. IDE version is 1.1.1 and CLI version is 2.8.6.

orgkhnargh, 2016-04-10:
Your `build` package installations seems to be corrupted. Please try to uninstall the `build` package (Preferences > Packages> build > Uninstall) and restart the IDE. The `build` package will be installed again after restart.

jakematlick, 2016-04-10:
That resolved it. Thanks!

closed 2016-04-10 by jakematlick