platformio/platformio-atom-ide #94: Failed to activate the platformio-ide package

micaelc, 2016-04-13:
[Enter steps to reproduce below:]

1. after installation 
2. rebotted pc 

**Atom Version**: 1.7.0
**System**: Microsoft Windows 10 Home Insider Preview
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.1.1

### Stack Trace

Failed to activate the platformio-ide package

```
At Cannot read property 'indexOf' of undefined

TypeError: Cannot read property 'indexOf' of undefined
    at Object.updateOSEnviron (C:/Users/micae/.atom/packages/platformio-ide/lib/maintenance.js:31:25)
    at Object.activate (C:/Users/micae/.atom/packages/platformio-ide/lib/main.js:52:17)
    at Package.module.exports.Package.activateNow (C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:183:20)
    at C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:156:32
    at Package.module.exports.Package.measure (C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:92:15)
    at C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:149:26
    at Package.module.exports.Package.activate (C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:146:34)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:538:21)
    at C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:519:29
    at Config.module.exports.Config.transactAsync (C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\config.js:337:18)
    at PackageManager.module.exports.PackageManager.activatePackages (C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:514:19)
    at PackageManager.module.exports.PackageManager.activate (C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:497:46)
    at C:\Users\micae\AppData\Local\atom\app-1.7.0\resources\app.asar\src\atom-environment.js:754:28
```

### Commands

```
```

### Config

```json
{}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.8.9
build, v0.59.0
language-ini, v1.16.0
linter, v1.11.4
linter-gcc, v0.6.9
platformio-ide, v1.1.1
tool-bar, v0.4.0

# Dev
No dev packages
```

n6nq, 2016-04-14:
I'm also having this issue. Pretty much the same circumstances. Windows 10 Home. Atom 1.7.0. Stack looks to be about the same.

TypeError: Cannot read property 'indexOf' of undefined
    at Object.updateOSEnviron (C:/Users/n6nq/.atom/packages/platformio-ide/lib/maintenance.js:31:25)
    at Object.activate (C:/Users/n6nq/.atom/packages/platformio-ide/lib/main.js:52:17)
    at Package.module.exports.Package.activateNow (C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:183:20)
    at C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:156:32
    at Package.module.exports.Package.measure (C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:92:15)
    at C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:149:26
    at Package.module.exports.Package.activate (C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:146:34)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:538:21)
    at C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:519:29
    at Config.module.exports.Config.transactAsync (C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\config.js:337:18)
    at PackageManager.module.exports.PackageManager.activatePackages (C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:514:19)
    at PackageManager.module.exports.PackageManager.activate (C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:497:46)
    at C:\Users\n6nq\AppData\Local\atom\app-1.7.0\resources\app.asar\src\atom-environment.js:754:28

viktak, 2016-04-14:
Same issue here. I have been using it for over a week with no problem. Even, this morning it started just fine. I had to restart the machine, and ever since I can't get it to work :(

[Enter steps to reproduce below:]

1. Launch Atom
2. ...

**Atom Version**: 1.7.0
**System**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.1.1

### Stack Trace

Failed to activate the platformio-ide package

```
At Cannot read property 'indexOf' of undefined

TypeError: Cannot read property 'indexOf' of undefined
    at Object.updateOSEnviron (C:/Users/Viktor.eromu/.atom/packages/platformio-ide/lib/maintenance.js:31:25)
    at Object.activate (C:/Users/Viktor.eromu/.atom/packages/platformio-ide/lib/main.js:52:17)
    at Package.module.exports.Package.activateNow (C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:183:20)
    at C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:156:32
    at Package.module.exports.Package.measure (C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:92:15)
    at C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:149:26
    at Package.module.exports.Package.activate (C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:146:34)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:538:21)
    at C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:519:29
    at Config.module.exports.Config.transactAsync (C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\config.js:337:18)
    at PackageManager.module.exports.PackageManager.activatePackages (C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:514:19)
    at PackageManager.module.exports.PackageManager.activate (C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:497:46)
    at C:\Users\Viktor.eromu\AppData\Local\atom\app-1.7.0\resources\app.asar\src\atom-environment.js:754:28
```

### Commands

```
```

### Config

```json
{
  "core": {},
  "platformio-ide": {}
}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.8.9
build, v0.59.0
language-ini, v1.16.0
linter, v1.11.4
linter-gcc, v0.6.9
platformio-ide, v1.1.1
platformio-ide-terminal, v2.0.3
tool-bar, v0.4.0

# Dev
No dev packages
```

ivankravets, 2016-04-14:
The problem is caused by new Atom 1.7.0 under Windows. Please use previous [stable version 1.6.2](https://github.com/atom/atom/releases/tag/v1.6.2) till we will fix issue.

viktak, 2016-04-14:
How does this explain why it has been working fine until now? Was there a "silent" update that I was not aware of?

ivankravets, 2016-04-14:
@viktak I've just opened separate issue https://github.com/atom/atom/issues/11460

viktak, 2016-04-14:
Thanks Ivan, I'll keep an eye on the development.

ayan4m1, 2016-04-14:
@viktak Atom updates in the background using a library called Squirrel.

ayan4m1, 2016-04-14:
Simple workaround for me was to edit my environment variables and change the case of my "Path" variable to "PATH" - restart to ensure the env got cleaned and Atom 1.7.0 is working with PlatformIO. My guess is that this is primarily affecting Win10 users because the default casing of the PATH var has changed to Path in this version.

ivankravets, 2016-04-14:
@ayan4m1 

> My guess is that this is primarily affecting Win10 users because the default casing of the PATH var has changed to Path in this version

I have this problem under Windows 7 in VM. I'm working on the fix.

ivankravets, 2016-04-14:
> Likely commit atom/atom@8c53e25 was also required for this behavior to start manifesting.

I've just made hotfix to avoid this Atom's issue. However, we have problems with our PlatformIO Terminal too.

ayan4m1, 2016-04-14:
Heh, just noticed the terminal thing, was hoping it was just me... appreciate the quick response in any event. I can help with test/repro if you need it.

ivankravets, 2016-04-14:
@ayan4m1 thanks! Could you test this patch https://github.com/platformio/platformio-atom-ide/commit/2ef99acd3d5711d84ffac761747556bf4d357731 ?

P.S: I'm working on the idea how to fix Terminal. It's a more complicated problem...

zoulzubazz, 2016-04-14:
reverting back to 1.6.2 fixed it.
close atom. download and install old installer from (setup.exe) 
https://github.com/atom/atom/releases/tag/v1.6.2

You should be up and running again. Did not have to uninstall 1.7.

viktak, 2016-04-14:
Yes, I can confirm 1.6.2 works fine. I uninstalled 1.7 first, though, just in case....

ivankravets, 2016-04-14:
@viktak, @zoulzubazz, @ayan4m1, @n6nq, @micaelc 

Please re-test the patch described in https://github.com/platformio/platformio-atom-ide-terminal/issues/12#issuecomment-209907331

ayan4m1, 2016-04-14:
@ivankravets Changed my PATH back to Path, rebooted, ran Atom 1.7.0 in dev mode for this package - checked out the develop branch which contains the referenced commit and platformio initializes successfully.

ivankravets, 2016-04-14:
@ayan4m1 Great! But how about terminal? Does it work? Do you have installed Visual Studio in a system?

ayan4m1, 2016-04-14:
Heh, just about to be work time where I am :) I have VS2015 with a working C++ toolchain but I won't have time to test for a few hours.

ivankravets, 2016-04-14:
@ayan4m1 thanks! :+1: Let's wait for @viktak, @zoulzubazz, @n6nq, @micaelc

webguy16, 2016-04-14:
Hello, i wasn't sure if i was suppose to comment on the patch thread or not so ill just put this here... I replaced my lib/main.js with the one linked(https://raw.githubusercontent.com/platformio/platformio-atom-ide/develop/lib/main.js) and get the same "Failed to activate..." error with a different stack trace. I do not have Visual Studio installed.  

**Atom Version**: 1.7.0
**System**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.1.1

### Stack Trace

Failed to activate the platformio-ide package

```
At Unexpected token import

SyntaxError: Unexpected token import
    at Module._compile (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\native-compile-cache.js:77:42)
    at Object.defineProperty.value [as .js] (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:715:27)
    at Package.module.exports.Package.activateNow (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:173:16)
    at C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:156:32
    at Package.module.exports.Package.measure (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:92:15)
    at C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:149:26
    at Package.module.exports.Package.activate (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package.js:146:34)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:538:21)
    at C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:519:29
    at Config.module.exports.Config.transactAsync (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\config.js:337:18)
    at PackageManager.module.exports.PackageManager.activatePackages (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:514:19)
    at PackageManager.module.exports.PackageManager.activate (C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\package-manager.js:497:46)
    at C:\Users\*userName*\AppData\Local\atom\app-1.7.0\resources\app.asar\src\atom-environment.js:754:28
```

### Commands

```
```

### Config

```json
{
  "core": {}
}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.8.9
build, v0.59.0
language-ini, v1.16.0
linter, v1.11.4
linter-gcc, v0.6.9
platformio-ide, v1.1.1
platformio-ide-terminal, v2.0.3
tool-bar, v0.4.0

# Dev
No dev packages
```

ivankravets, 2016-04-14:
@webguy16 
> At Unexpected token import

Looks like you have broken file. Please update `lib/main.js` and check that you have copied all content.

webguy16, 2016-04-14:
Well don't i feel smart... I just right clicked and 'saved as' and replaced the file this time rather than 'select all - copy- select all - paste' and it looks to be working fine. Sorry bout that.

viktak, 2016-04-14:
I'm afraid, this fix only allows PlatformIO to start without an error, but the terminal and anything that uses the terminal freezes. I tried terminal, serial port monitor, list serial ports and they all freeze.

ivankravets, 2016-04-14:
Friends, we are working on it... The problem is not linked only with https://github.com/atom/atom/issues/11302

They also updated `node.js` that broke Pty.JS. 

I can't imagine who allowed in Atom's maintainer group to release 1.7.0 with these issues.

ayan4m1, 2016-04-14:
I have gotten the terminal working again in a dev environment. If you apm rebuild with the msbuild toolchain, you can get pty.js working again. More info when I have time to write it up.

ivankravets, 2016-04-14:
@ayan4m1 Could you share with us `~/.atom/packages/platformio-ide-terminal/node_modules/pty.js/build/Release/pty.node` file? Please zip it.

ayan4m1, 2016-04-14:
[pty.zip](https://github.com/platformio/platformio-atom-ide/files/219346/pty.zip)
Enjoy :)

ivankravets, 2016-04-14:
> Enjoy :)

It works for my Win7 in VM. Thanks.

viktak, 2016-04-14:
Can you tell me how to test it? I looked up my  ~/.atom/packages/platformio-ide-terminal/node_modules/pty.js/build/Release folder, and it's empty.

viktak, 2016-04-14:
In the process of trying to find out more about this issue, I removed Atom from my PC, including the .atom directory. Then I went ahead and installed the Atom 1.6.2 and PlatformIO from scratch. Once all was finished I tried to launch a terminal (or anything else that uses the terminal window) and I get the following error message:

[image]

As you can see on the screenshot, the package is clearly installed.

Anybody has an idea what might be wrong? Since this morning things have been going from bad to worse...

viktak, 2016-04-14:
@ivankravets I think this is because you are preparing the new version, 2.0.4.

ivankravets, 2016-04-14:
I've just release PlatformIO Terminal 2.0.5 - please re-test it. If it works, I'll release IDE. Thanks.

viktak, 2016-04-14:
Downloading now, keep fingers crossed...

ivankravets, 2016-04-14:
@viktak looks like it also doesn't work.

I've tried a lot of solutions today  and can't find single which will work on both Atoms: 1.6 & 1.7.

viktak, 2016-04-14:
@ivankravets I'm afraid still the same issue. Tried it on Atom 1.7:
[image]

viktak, 2016-04-14:
@ivankravets Just tried it also in Atom 1.6.2, same problem

ivankravets, 2016-04-14:
@viktak do you see "red" :bug: in the status bar? Bottom.

viktak, 2016-04-14:
@ivankravets Yes, now that you say it :) 
This is what I can see in the details:
`pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node
pty.js@0.3.0 – %1 is not a valid Win32 application.
 \\?\C:\Users\Viktor.eromu\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32.node`

viktak, 2016-04-14:
Rebuilt it, and now I am back to the stage whene the terminal pane freezes on launch :(

ivankravets, 2016-04-14:
Yessssss! I've just won it! I forked https://github.com/platformio/pty.js and forced it to work for Atom 1.6/1.7.

Pelase re-test the latest PlatformIO IDE Terminal 2.0.8

viktak, 2016-04-14:
@ivankravets Yessssss! you made it! I managed to get it to work with Atom 1.7 and terminal 2.0.9. I now how to go to bed, but tomorrow I'll double check it on my main machine too.

Well done and thanks for all your efforts, Ivan! If there is anything I can help you with (mainly testing), give me a shout!

viktak, 2016-04-14:
@ivankravets ok, I can now confirm: 2.0.9 works with 1.7, but only if the patched lib/main.js is used.
Now, it's really time to sleep...

ivankravets, 2016-04-14:
@viktak thanks for the testing. PlatformIO IDE 1.1.2 has not been released yet. I'm working on it.

ivankravets, 2016-04-14:
>  If there is anything I can help you with (mainly testing), give me a shout!

Please give us a star :star2: 
- https://github.com/platformio/platformio-atom-ide/stargazers
- https://github.com/platformio/platformio/stargazers

Thanks in advance!

n6nq, 2016-04-15:
Thanks so much for your fast action  on this issue. 

As a gray beard programmer,  it's always impressive to see dedication and enthusiasm. 

As a newbie to PlatformIO,  it is heartening and encouraging your kick butt.

Thanks again,

Jon

N6NQ

ivankravets, 2016-04-15:
@n6nq Thanks a lot for the warm words!

/all, I close this issue. [PlatformIO IDE 1.1.2 is released](https://community.platformio.org/t/platformio-ide-1-1-2/153).

If you have any problems please open new issue.

closed 2016-04-15 by ivankravets

viktak, 2016-04-15:
> Please give us a star :star2: 
Done and done.

I just upgraded both ide and terminal to the latest versions and all went smoothly.

Life is good again! :)

micaelc, 2016-04-15:
same here !!! sun is shining again !!!

sk1tt1sh, 2016-06-21:
(Windows) Seems like running Atom as administrator makes that error go away. Is something writing a file in the %programfiles% directory or anywhere other than %appdata%?

sdaitzman, 2017-05-24:
I have raised this issue again at #493 because I'm having it. This comment is to link the issues.