platformio/platformio-atom-ide #95: Failed to load the platformio-ide package

jmscaramal, 2016-04-14:
Just installed atom-editor and platformio-ide next. So, after the installation, I get this error and I can't load the platformio-ide package.

Any help?

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.7.0
**System**: linux 4.4.5-1-ARCH
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.1.1

### Stack Trace

Failed to load the platformio-ide package

```
At ENOENT, common/api/lib/shell.js not found in /usr/share/atom/resources/atom.asar

Error: ENOENT, common/api/lib/shell.js not found in /usr/share/atom/resources/atom.asar
    at notFoundError (ATOM_SHELL_ASAR.js:110:13)
    at Object.fs.readFileSync (ATOM_SHELL_ASAR.js:435:9)
    at compileFileAtPath (/usr/share/atom/resources/app.asar/src/compile-cache.js:69:23)
    at Object.defineProperty.value [as .js] (/usr/share/atom/resources/app.asar/src/compile-cache.js:207:18)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:50:27)
    at Object.<anonymous> (/home/scaramal/.atom/packages/platformio-ide/lib/main.js:21:24)
    at Module._compile (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/usr/share/atom/resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/usr/share/atom/resources/app.asar/src/package.js:715:27)
    at /usr/share/atom/resources/app.asar/src/package.js:117:28
    at Package.module.exports.Package.measure (/usr/share/atom/resources/app.asar/src/package.js:92:15)
    at Package.module.exports.Package.load (/usr/share/atom/resources/app.asar/src/package.js:106:12)
    at PackageManager.module.exports.PackageManager.loadPackage (/usr/share/atom/resources/app.asar/src/package-manager.js:457:14)
    at /usr/share/atom/resources/app.asar/node_modules/settings-view/lib/package-manager.js:454:29
    at exit (/usr/share/atom/resources/app.asar/node_modules/settings-view/lib/package-manager.js:73:16)
    at triggerExitCallback (/usr/share/atom/resources/app.asar/src/buffered-process.js:215:47)
    at /usr/share/atom/resources/app.asar/src/buffered-process.js:229:18
    at Socket.<anonymous> (/usr/share/atom/resources/app.asar/src/buffered-process.js:100:18)
    at emitOne (events.js:82:20)
    at Socket.emit (events.js:169:7)
    at Pipe._onclose (net.js:469:12)
```

### Commands

```
     -0:34.6.0 core:confirm (atom-text-editor.editor.mini.is-focused)
```

### Config

```json
{
  "core": {}
}
```

### Installed Packages

```coffee
# User
platformio-ide, v1.1.1

# Dev
No dev packages
```

orgkhnargh, 2016-04-14:
Seems like `shell` has been moved to `electron` module. Related Atom commit: https://github.com/atom/atom/commit/1725b9bf54252be2a2cf81d0b76bd5cb1693d918

I just noticed that ApplicationDelegate exposes the `openExternal` [here](https://github.com/atom/atom/blob/1725b9bf54252be2a2cf81d0b76bd5cb1693d918/src/application-delegate.coffee#L197), so it's now possible to open external URLs via `atom.applicationDelegate.openExternal(url)` (instead of `shell.openExternal(url)`). The problem is that I can't find any docs about ApplicationDelegate, so we can't rely on it on a long term, I guess.

orgkhnargh, 2016-04-14:
@jmscaramal sorry for the inconvenience! Can you please stick with Atom 1.6 until we fix all 1.7 related issues? Atom releases are available to download from here: https://github.com/atom/atom/releases

jmscaramal, 2016-04-14:
Thanks for replying so fast @orgkhnargh. I've downgraded atom to version 1.6.2 and it worked like a charm!

ivankravets, 2016-04-15:
@jmscaramal PlatformIO 1.1.2 has been released and works with Atom 1.7. https://community.platformio.org/t/platformio-ide-1-1-2/153

Do you still have this problem?

jmscaramal, 2016-04-15:
Hello @ivankravets! I've updated just now and it worked! Thanks for alerting me!

closed 2016-04-15 by jmscaramal

Nasaki, 2016-05-28:
just new platformio. here error i got after installed the package.
Could anyone advise and help?

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.8.0-beta4
**System**: "Manjaro Linux"
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.2.2

### Stack Trace

Failed to load the platformio-ide package

```
At ENOENT, common/api/lib/shell.js not found in /usr/local/share/atom-beta/resources/atom.asar

Error: ENOENT, common/api/lib/shell.js not found in /usr/local/share/atom-beta/resources/atom.asar
    at notFoundError (ATOM_SHELL_ASAR.js:110:13)
    at Object.fs.readFileSync (ATOM_SHELL_ASAR.js:435:9)
    at compileFileAtPath (/usr/local/share/atom-beta/resources/app.asar/src/compile-cache.js:69:23)
    at Object.defineProperty.value [as .js] (/usr/local/share/atom-beta/resources/app.asar/src/compile-cache.js:207:18)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/usr/local/share/atom-beta/resources/app.asar/src/native-compile-cache.js:50:27)
    at Object.<anonymous> (/home/psuthipa/.atom/packages/platformio-ide/lib/maintenance.js:25:41)
    at Module._compile (/usr/local/share/atom-beta/resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/usr/local/share/atom-beta/resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/usr/local/share/atom-beta/resources/app.asar/src/native-compile-cache.js:50:27)
    at Object.<anonymous> (/home/psuthipa/.atom/packages/platformio-ide/lib/main.js:22:20)
    at Module._compile (/usr/local/share/atom-beta/resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/usr/local/share/atom-beta/resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/usr/local/share/atom-beta/resources/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/usr/local/share/atom-beta/resources/app.asar/src/package.js:715:27)
    at /usr/local/share/atom-beta/resources/app.asar/src/package.js:117:28
    at Package.module.exports.Package.measure (/usr/local/share/atom-beta/resources/app.asar/src/package.js:92:15)
    at Package.module.exports.Package.load (/usr/local/share/atom-beta/resources/app.asar/src/package.js:106:12)
    at PackageManager.module.exports.PackageManager.loadPackage (/usr/local/share/atom-beta/resources/app.asar/src/package-manager.js:457:14)
    at /usr/local/share/atom-beta/resources/app.asar/node_modules/settings-view/lib/package-manager.js:454:29
    at exit (/usr/local/share/atom-beta/resources/app.asar/node_modules/settings-view/lib/package-manager.js:73:16)
    at triggerExitCallback (/usr/local/share/atom-beta/resources/app.asar/src/buffered-process.js:215:47)
```

### Commands

```
     -1:02.8.0 core:confirm (atom-text-editor.editor.mini.is-focused)
```

### Config

```json
{
  "core": {
    "disabledPackages": [
      "platformio-ide-terminal",
      "platformio-aia",
      "language-arduino",
      "arduino-toolbelt",
      "mcduino"
    ]
  }
}
```

### Installed Packages

```coffee
# User
platformio-ide, v1.2.2

# Dev
No dev packages
```

jourjine, 2016-07-21:
[Enter steps to reproduce below:]

1. Import Arduino project for esp8266
2. build
3. have error
4. close, deinstalling, cleaning, install Atom again
5. on start always this error

**Atom Version**: 1.8.0
**System**: Mac OS X 10.11.5
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.3.4

### Stack Trace

Failed to load the platformio-ide package

```
At Cannot find module 'fs-extra'

Error: Cannot find module 'fs-extra'
    at Module._resolveFilename (module.js:338:15)
    at Function.Module._resolveFilename (/Applications/Atom.app/Contents/Resources/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:289:25)
    at Module.require (module.js:366:17)
    at require (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:50:27)
    at module.exports (/Users/KOCMOHABT/.atom/packages/build-platformio/node_modules/promisify-node/index.js:157:15)
    at Object.<anonymous> (/Users/KOCMOHABT/.atom/packages/platformio-ide/lib/project-examples/command.js:27:13)
    at Module._compile (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/Applications/Atom.app/Contents/Resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:50:27)
    at Object.<anonymous> (/Users/KOCMOHABT/.atom/packages/platformio-ide/lib/main.js:37:66)
    at Module._compile (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/Applications/Atom.app/Contents/Resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:715:27)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/package.js:117:28
    at Package.module.exports.Package.measure (/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:92:15)
    at Package.module.exports.Package.load (/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:106:12)
    at PackageManager.module.exports.PackageManager.loadPackage (/Applications/Atom.app/Contents/Resources/app.asar/src/package-manager.js:457:14)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/package-manager.js:402:19
    at Config.module.exports.Config.transact (/Applications/Atom.app/Contents/Resources/app.asar/src/config.js:312:16)
    at PackageManager.module.exports.PackageManager.loadPackages (/Applications/Atom.app/Contents/Resources/app.asar/src/package-manager.js:397:19)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/atom-environment.js:743:28
```

### Commands

```
```

### Config

```json
{
  "core": {
    "disabledPackages": [
      "language-gfm"
    ]
  }
}
```

### Installed Packages

```coffee
# User
build, v0.65.0
build-platformio, v1.2.2
busy, v0.5.0
language-arduino, v0.4.1
language-ino, v0.1.0
language-markdown, v0.15.0
language-todo-extra-words, v0.2.0
platformio-ide, v1.3.4
platformio-ide-terminal, v2.1.1
terminal-plus, v0.14.5

# Dev
No dev packages
```

ivankravets, 2016-07-21:
@jourjine do you see `fs-extra` folder here `~/.atom/packages/platformio-ide/node_modules/`? 

If no, please `rm -rf ~/.atom/packages/platformio-ide` and restart Atom.

jourjine, 2016-07-21:
i see `fs-extra` in folder, but have error.
i`m delete all in atom folder and restart Atom. This solve all problem

ivankravets, 2016-07-21:
@jourjine thanks for the report.

fabalcu97, 2016-07-24:
Hi, I hope somebody can help me, I've installed PlatformIO-IDE recently and got this issue.

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.8.0
**System**: linux 4.4.6-gentoo
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.3.5

### Stack Trace

Failed to load the platformio-ide package

```
At ENOENT, common/api/lib/shell.js not found in /usr/lib64/electron/resources/atom.asar

Error: ENOENT, common/api/lib/shell.js not found in /usr/lib64/electron/resources/atom.asar
    at notFoundError (ATOM_SHELL_ASAR.js:110:13)
    at Object.fs.readFileSync (ATOM_SHELL_ASAR.js:435:9)
    at compileFileAtPath (/usr/lib64/atom/app.asar/src/compile-cache.js:69:23)
    at Object.defineProperty.value [as .js] (/usr/lib64/atom/app.asar/src/compile-cache.js:207:18)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/usr/lib64/atom/app.asar/src/native-compile-cache.js:50:27)
    at Object.<anonymous> (/home/fabricio/.atom/packages/platformio-ide/lib/main.js:20:24)
    at Module._compile (/usr/lib64/atom/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/usr/lib64/atom/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/usr/lib64/atom/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/usr/lib64/atom/app.asar/src/package.js:715:27)
    at /usr/lib64/atom/app.asar/src/package.js:117:28
    at Package.module.exports.Package.measure (/usr/lib64/atom/app.asar/src/package.js:92:15)
    at Package.module.exports.Package.load (/usr/lib64/atom/app.asar/src/package.js:106:12)
    at PackageManager.module.exports.PackageManager.loadPackage (/usr/lib64/atom/app.asar/src/package-manager.js:457:14)
    at /usr/lib64/atom/app.asar/src/package-manager.js:402:19
    at Config.module.exports.Config.transact (/usr/lib64/atom/app.asar/src/config.js:312:16)
    at PackageManager.module.exports.PackageManager.loadPackages (/usr/lib64/atom/app.asar/src/package-manager.js:397:19)
    at /usr/lib64/atom/app.asar/src/atom-environment.js:743:28
```

### Commands

```
```

### Config

```json
{
  "core": {
    "autoHideMenuBar": true,
    "disabledPackages": [
      "server-script"
    ],
    "themes": [
      "one-dark-ui",
      "atom-dark-syntax"
    ]
  }
}
```

### Installed Packages

```coffee
# User
platformio-ide, v1.3.5
platformio-ide-terminal, v2.1.1
processing-autocomplete, v0.2.0
terminal-plus, v0.14.5

# Dev
No dev packages
```

sledgeh, 2016-08-28:
Linux 4.1.31-1-MANJARO
Atom v1.9.9
platformio-ide v1.4.0

```
Cannot find module 'fs-extra'
Hide Stack Trace
Error: Cannot find module 'fs-extra'
    at Module._resolveFilename (module.js:339:15)
    at Function.Module._resolveFilename (/usr/share/atom/resources/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:290:25)
    at Module.require (module.js:367:17)
    at require (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:50:27)
    at module.exports (/home/kyb/.atom/packages/build-platformio/node_modules/promisify-node/index.js:157:15)
    at Object.<anonymous> (/home/kyb/.atom/packages/platformio-ide/lib/project-examples/command.js:27:13)
    at Module._compile (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/usr/share/atom/resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:50:27)
    at Object.<anonymous> (/home/kyb/.atom/packages/platformio-ide/lib/main.js:37:66)
    at Module._compile (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/usr/share/atom/resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (/usr/share/atom/resources/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/usr/share/atom/resources/app.asar/src/package.js:715:27)
    at /usr/share/atom/resources/app.asar/src/package.js:117:28
    at Package.module.exports.Package.measure (/usr/share/atom/resources/app.asar/src/package.js:92:15)
    at Package.module.exports.Package.load (/usr/share/atom/resources/app.asar/src/package.js:106:12)
    at PackageManager.module.exports.PackageManager.loadPackage (/usr/share/atom/resources/app.asar/src/package-manager.js:457:14)
    at PackageManager.module.exports.PackageManager.activatePackage (/usr/share/atom/resources/app.asar/src/package-manager.js:536:30)
    at /usr/share/atom/resources/app.asar/node_modules/settings-view/lib/package-manager.js:452:29
    at exit (/usr/share/atom/resources/app.asar/node_modules/settings-view/lib/package-manager.js:73:16)
    at triggerExitCallback (/usr/share/atom/resources/app.asar/src/buffered-process.js:215:47)
    at /usr/share/atom/resources/app.asar/src/buffered-process.js:222:18
    at Socket.<anonymous> (/usr/share/atom/resources/app.asar/src/buffered-process.js:100:18)
    at emitOne (events.js:95:20)
    at Socket.emit (events.js:182:7)
    at Pipe._onclose (net.js:477:12)
```

ivankravets, 2016-08-28:
@sledgeh try `rm -rf ~/.atom/packages/platformio-ide` and restart Atom. Install plugin again.

sledgeh, 2016-08-29:
@ivankravets no effect. Same error.