platformio/platformio-atom-ide #96: Serial Monitor not showing anything

micaelc, 2016-04-15:
Windows 10 Home
Atom 1.7.0
Platformio 1.1.2

i can do everything (Build / Upload / ... )
i can open the Serial Monitor 
but nothing happens !! 

i use also Mac environment and there i am able to use Serial monitor with no problem

any help ???

micaelc, 2016-04-15:
Hello again...
it seams that the CLI is not working
when i open the terminal / serial monitor nothing happens (black panel opens... with only blinking carret)

Can anyone help me on this !!

ivankravets, 2016-04-16:
> when i open the terminal / serial monitor nothing happens (black panel opens... with only blinking carret)

Please open `Menu: View > Developer > Toggle Developer Tools` and dump here the output from the console.

Thanks.

micaelc, 2016-04-16:
This is solved...
under Win 10 you can't have a username and the path to your projects with spaces or special characters.
after taking care of that... Platformio is back to normal...

thanks anyway...

ivankravets, 2016-04-17:
Thanks for the report.

closed 2016-04-17 by ivankravets

reza7rm, 2017-05-23:
I am having the same problem and the terminal and serial monitor show a windows powershell in which I can't do anything!
Win 10
Core 3.4.0b3
IDE 2.0.0-beta.3
[image]

its-Hashim, 2018-10-09:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.31.1 x64
**Electron**: 2.0.7
**OS**: Microsoft Windows 10 Home Single Language
**Thrown From**: [platformio-ide-terminal](https://github.com/platformio/platformio-atom-ide-terminal) package 2.8.4

### Stack Trace

Uncaught channel closed

```
At events.js:183

Error [ERR_IPC_CHANNEL_CLOSED]: channel closed
    at ChildProcess.target.send (internal/child_process.js:588:16)
    at Task.module.exports.Task.send (~/AppData/Local/atom/app-1.31.1/resources/app/src/task.js:113:33)
    at PlatformIOTerminalView.module.exports.PlatformIOTerminalView.resize (/packages/platformio-ide-terminal/lib/view.coffee:276:17)
    at PlatformIOTerminalView.module.exports.PlatformIOTerminalView.resizeTerminalToView (/packages/platformio-ide-terminal/lib/view.coffee:503:6)
    at PlatformIOTerminalView.module.exports.PlatformIOTerminalView.focus (/packages/platformio-ide-terminal/lib/view.coffee:465:6)
    at PlatformIOTerminalView.focus (/packages/platformio-ide-terminal/lib/view.coffee:1:1)
    at /packages/platformio-ide-terminal/lib/status-bar.coffee:105:25
```

### Commands

```
     -8:46.3.0 platformio-ide:target:build (input.hidden-input)
     -8:46.3.0 platformio-ide:target:build-F:\tecki\temp\temp (atom-workspace.workspace.scrollbars-visible-always.theme-atom-light-syntax.theme-atom-light-ui)
     -8:17.1.0 platformio-ide:target:build (input.hidden-input)
     -8:17.1.0 platformio-ide:target:build-F:\tecki\temp\temp (atom-workspace.workspace.scrollbars-visible-always.theme-atom-light-syntax.theme-atom-light-ui)
     -8:00.5.0 platformio-ide:target:upload (input.hidden-input)
     -8:00.5.0 platformio-ide:target:upload-F:\tecki\temp\temp (atom-workspace.workspace.scrollbars-visible-always.theme-atom-light-syntax.theme-atom-light-ui)
     -7:43.2.0 project-find:show (input.hidden-input)
     -7:38.2.0 platformio-ide:maintenance.open-terminal (input.hidden-input)
     -7:22.9.0 application:show-settings (atom-workspace.workspace.scrollbars-visible-always.theme-atom-light-syntax.theme-atom-light-ui)
  2x -6:17.1.0 platformio-ide:maintenance.serial-monitor (input.hidden-input)
```

### Non-Core Packages

```
autocomplete-clang 0.11.5 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.25 
intentions 1.1.5 
language-ini 1.20.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.7.1 
platformio-ide 2.2.2 
platformio-ide-debugger 1.2.6 
platformio-ide-terminal 2.8.4 
tool-bar 1.1.10 
```

the above error i get when i open serial monitor in platform io
please somebody help