platformio/platformio-atom-ide #97: Installation of PlatformIO fails

RREE, 2016-04-17:
Recently I came across Atom and PlatformIO. So wanted to to give it a try.

I installed the latest Atom as of today (2016-04-17). Then I started to install platformio. It first complained about missing Python (although I already had V2.7.11 on my computer). I reconfigured Python to include to correct PATH and restarted the installation. 

I removed first only C:\Users\Rolf\.atom, tried again, than I removed completely Atom from the computer (via control panel) and started again to no avail. Any hints of what I am doing wrong?

best regards
   RE

[image]

orgkhnargh, 2016-04-18:
Hello @RREE,

Can you please replace the `C:\Users\Rolf\.atom\packages\platformio-ide\lib\install\command.js` file with this one: https://github.com/platformio/platformio-atom-ide/raw/31ab855b54449e3fabb064e77b0ad879be5545c4/lib/install/command.js
This should give us more information about the cause of the error.

ivankravets, 2016-06-14:
Please reopen if you still need any help

closed 2016-06-14 by ivankravets