platformio/platformio-atom-ide #99: Upgrade from 1.1.1 to 1.1.2 fails on first attempt (minor)

jbbr, 2016-04-18:
I tried upgrading platformio-ide from 1.1.1 to 1.1.2 and got the following error in the dev console:
```
Uncaught Error: spawn /Users/user/.atom/packages/platformio-ide/penv/bin/python ENOENT
```
According to the console error is in `events.js:141`
The directory ` /Users/user/.atom/packages/platformio-ide/penv` does not exists at all (Although everything worked fine before upgrading - I assume it has been deleted during upgrade process)

Cancelling "Installing platformio-ide" dialog did not worked as well.
What did worked was quitting Atom and restarting it. The "Installing platformio-ide" dialog showed again and this time finished successfully.
So this is not a blocker but still confusing and not a perfect experience.

Atom version 1.6.0 (Installed with official platformio-atom-ide installer)
System: OSX 10.11

orgkhnargh, 2016-04-21:
I haven't been able to reproduce this error.

jbbr, 2016-04-28:
Just had this problem again when upgrading from 1.1.2 to 1.2.0.

- relatively fresh install of platformio-ide plugin in Atom 1.7.2 (This time not official platformio-atom-ide installer - used "brew cask" instead)
- pressed update in settings for platformio-ide

So nothing fancy here. I made a screenshot this time:

[image]

Btw: When pressing "Cancel" the button switches to a disabled "Canceling" button but canceling does not work.
[image]

Is there anything I can do to find the cause of this issue?

jbbr, 2016-04-28:
Just restarted Atom and this time I got:
[image]

Update of the plugin itself did worked according to atom (Settings/plugins shows 1.2.0). Also the platformio toolbar is visible. But:

When clicking on "Build" button execution of platformio command failes:

[image]

The directory `~/.atom/packages/platformio-ide/penv` is missing. In previous versions of platformio-ide it got recreated automatically when starting atom. This does not seem to work anymore because it seems to think that an installation is still running (Maybe there is a lock file or something which hasn't been deleted due to broken "canceling"?)

ivankravets, 2016-04-29:
We have release PlatformIO IDE 1.2.1 today. Do you still have this problem?

jbbr, 2016-04-29:
Much better now:
- Opened Atom and pressed "Upgrade all" in Plugins section
- Got "PlatformIO installing" message which finished successfully
- After finish building any project failed with some Atom-warning toast (Sorry, I forgot the message)
- Restarted Atom, got again the "PlatformIO installing" message, now it asked me to reload the IDE which I did
=> Since then everything is fine

So upgrade experience was a bit rough but at least it worked well in the end.

Thanks!

ivankravets, 2016-06-14:
Please reopen if you still need any help

closed 2016-06-14 by ivankravets